PX4/PX4-ECL #516: Fusing 'indoor GPS'

Seanmatthews, 2018-10-29:
Should there be another sensor measurement added for indoor sensor types, like ultra-wideband radio ToF positioning, that resemble GPS in its error estimates and GDoP? If not, would MoCap work for such a sensor? Or, would it be best to convert XYZ to a fake GPS coordinate and use the GPS measurement channel to insert the measurement into EKF2?

I'm happy to work on adding a new measurement type if that's what's required, but I'd rather not create unnecessary work if one of the other topics will suffice.

mhkabir, 2018-10-29:
> that resemble GPS in its error estimates and GDoP

Bad assumption. UWB positioning for example, generally requires fusion of raw tag ranges in the navigation EKF for good performance. PX4's EKF2 doesn't have an interface for this, but I believe @priseborough derived the equations and had a baseline interface going for Ardupilot at some point.

Seanmatthews, 2018-10-30:
@TSC21 Yes, position control in offboard. Follow-up question about the feeding the measurement in as a vision input, and forgive me that this touches on MAVROS as well (@mhkabir please chime in on this as well)-- I'm feeding in a vision pose estimate to `~vision_pose/pose`. If I feed in the UWB position measurement to `~mocap/pose`, it's not clear to me in the EKF2_AID_MASK how those are handled separately in ecl EKF2. Can EKF2 handle both of those topics coming from MAVROS?

@mhkabir I'm going on information from the manufacturer regarding Decawave GDoP.

TSC21, 2018-10-30:
No. You need to use the vision topic so to feed mocap data to EKF2.

Seanmatthews, 2018-10-30:
@TSC21 Thanks for clarifying this.

@priseborough Can you comment on the baseline interface that you potentially created for Ardupilot? Do you think it's somewhat portable to PX4?

priseborough, 2018-10-30:
The PX4/ecl/EKF and ArduPilot/AP_NavEKF3 libraries use the same state vector and mathematical derivation so same observation equations for range measurements can therefore be used by PX4 (but not the code due to licensing issues)

Two problems were encountered when using this method:

1) The ambiguity associated with co-planar beacons - you have to constrain the solution to one side of the plane the beacons are positioned on or, in the case of ArduPilot, implement a dual hypothesis filter.

2) Numerical instability when very close to a beacon - this required some ad-hoc work arounds.

priseborough, 2021-08-17:
We currently support external positioning systems (also velocity and quaternions) using local coordinates via the vehicle_odometry message. For global coordinates, there is the vehicle_gps message.

closed 2021-08-17 by priseborough