PX4/PX4-ECL #519: Issues with heading estimation

CarlOlsson, 2018-11-09:
@priseborough 
There seems to be serious issues present regarding heading estimation at places with high magnetic declination.

The problem is easy to reproduce in SITL. Just [change the location to e.g. Seattle](https://github.com/PX4/sitl_gazebo/blob/3d80f63562c24c1537e1f8423ce649c99ebc15ea/include/gazebo_gps_plugin.h#L97-L100) with a magnetic declination of ~15 degrees, ```make clean```, run ```make posix_sitl_default gazebo``` and fly a mission.

Here is a [logfile](https://logs.px4.io/plot_app?log=ffe9fb87-6a4b-44ae-b781-d58ad959e969) from such a case. I also set EKF2_MAG_DECL to -5 degrees for illustration purpose but it is not needed to expose the bug.

#### The following happens:

t = t0
The system boots up. The true heading is 25 degrees and the true declination is 15 degrees.
The EKF starts to initialize the filter by collecting and low pass filtering the magnetometer and barometer sensor data.

t = t1
When the EKF has collected 2 times the observation buffer length it proceeds with the initialization by running the [resetMagHeading](https://github.com/PX4/ecl/blob/master/EKF/ekf_helper.cpp#L546) function. Here it initialize the heading by taking the last magnetometer measurement, rotate it into the inertial frame using the current attitude estimate ignoring the yaw, calculate the angle between the N and E component and add the magnetic declination. However, the declination is taken from the class variable _mag_declination and since the function [calcMagDeclination()](https://github.com/PX4/ecl/blob/master/EKF/ekf_helper.cpp#L753) has not yet run a single time it will use the [default value of zero](https://github.com/PX4/ecl/blob/master/EKF/ekf.h#L375).

Now the function [initialiseFilter()](https://github.com/PX4/ecl/blob/master/EKF/ekf.cpp#L135) returns true and the function [controlFusionModes()](https://github.com/PX4/ecl/blob/10a0fef7b7db749f5baf3384e9c900d3aa0e8140/EKF/control.cpp#L46-L46) is called. First thing that happens here is that we call [calcMagDeclination()](https://github.com/PX4/ecl/blob/10a0fef7b7db749f5baf3384e9c900d3aa0e8140/EKF/ekf_helper.cpp#L753-L753). In here since _NED_origin_initialised is false we set the _mag_declination variable to the EKF2_MAG_DECL parameter, in our case -5.
After that we start to wait for the tilt alignment and when that is done we again run the function [resetMagHeading](https://github.com/PX4/ecl/blob/master/EKF/ekf_helper.cpp#L546).
This time the _mag_declination is set to EKF2_MAG_DECL and the heading estimate is still zero so the earth magnetic field states are set to correspond to a declination of -5 degrees.

Now both tilt and yaw alignment is finished and the EKF is up and running. Since the heading is estimated to be zero degrees and the _mag_declination is using the parameter value the measured heading from the magnetometer will correspond to 10 degrees (true heading is 25 deg, true declination is 15 deg). The innovations will thus start at 10 degrees and slowly decrease simultaneously to the estimated heading rotating to 10 degrees. The estimated heading will even overshoot since the gyro z bias will grow.

t = t2
Now the EKF gets GPS lock and sets _NED_origin_initialised to true. In the next iteration of [calcMagDeclination()](https://github.com/PX4/ecl/blob/10a0fef7b7db749f5baf3384e9c900d3aa0e8140/EKF/ekf_helper.cpp#L753-L753) it then updates the _mag_declination variable to the value returned by the look up table on the GPS position. _mag_declination is hence set to the true value of 15 degrees. Now the heading innovation starts at 15 degrees, the estimated heading will start to rotate against the true value of 25 but will overshoot massively because of the now very large gyro Z bias. 

t = t3
The quad takes off and starts flying the mission with a **heading estimate error of 10 degrees**. 

t = t4
The quad finished the mission. The heading estimate has now converged to the true value

[image]

Heading innovations
[image]

blue is the magnetic declination calculated in postprocessing from the estimator states, red is the ecl class variable _mag_declination added to logging
[image]

CarlOlsson, 2018-11-09:
Flying almost the same mission but with EKF2_MAG_TYPE = 0 (automatic switching from 3D axes and simple heading fusion) is even worse. 

This is the true heading, estimated heading, the magnetic declination calculated from the estimated earth magnetic field states and the EKF2_MAG_DECL parameter added to logging
[image]

Note that the second time the exact same mission is flown everything is fine since the EKF2_MAG_DECL automatically is updated to the correct value in the first flight
[image]

dagar, 2018-11-09:
As a first step what about grabbing the declination in resetMagHeading? At least then it would be restored if previously saved to the parameter.

This still doesn't on a completely fresh system or the first use in a new area. What about forcing realignYawGPS() as soon as the GPS checks pass and NED origin is initialized?

 - https://github.com/PX4/ecl/tree/pr-ekf_mag_decl_cleanup
 - https://github.com/PX4/Firmware/tree/pr-ekf_mag_decl_cleanup

CarlOlsson, 2018-11-12:
> @CarlOlsson If you're using heading fusion you are directly extracting the measured heading from the magnetometer measurement. By definition there are no magnetic earth field states involved.

The estimated earth magnetic field states could still be used to calculate a more accurate value of the magnetic declination than the one obtained from the GEO lookup table. Which, actually is [already done](https://github.com/PX4/ecl/blob/10a0fef7b7db749f5baf3384e9c900d3aa0e8140/EKF/ekf_helper.cpp#L758), just missed it before

priseborough, 2018-11-12:
I like the idea of a running realignYawGPS() as soon as the WGS-84 origin has been set. Will submit a patch for that.

RomanBapst, 2018-11-12:
> The estimated earth magnetic field states could still be used to calculate a more accurate value of the magnetic declination than the one obtained from the GEO lookup table. Which, actually is already done, just missed it before
@CarlOlsson That's true as long as your not using heading fusion only. In that case the mag alignment never takes place.

priseborough, 2018-11-12:
The resetMagHeading() function currently does not reset the states and covariances unless the change in magnetic yaw is greater than 15 degrees. This was done originally to avoid unnecessary disturbance of the covariance matrix when doing the in-air yaw alignment. As a compromise we could reset the yaw angle when the change is <15 degrees, but leave the covariance matrix alone.

priseborough, 2018-11-12:
See https://github.com/PX4/ecl/pull/520

closed 2018-11-14 by priseborough