PX4/PX4-ECL #522: Possible issue with saving mag decl parameter

CarlOlsson, 2018-11-20:
There might be a possibility that when ekf2_main.cpp wants to save the magnetic declination returned by the GEO lookup table it instead save the value 0.

This happens since

get_mag_decl_deg() returns true only when _NED_origin_initialised is true
https://github.com/PX4/ecl/blob/c49ab760c78fc6dc4cb174d6f9b7b39b01e19a7b/EKF/estimator_interface.h#L338-L342

_NED_origin_initialised is set to true in collect_gps()
https://github.com/PX4/ecl/blob/c49ab760c78fc6dc4cb174d6f9b7b39b01e19a7b/EKF/gps_checks.cpp#L78-L81

but the _mag_declination_to_save_deg is only updated in calcMagDeclination() which is only run in controlFusionModes() which only runs if _imu_updated is true 

So if the GPS checks pass and update() runs with _imu_updated false then the mag decl parameter will be saved as zero.

I haven't seen this happening in a real flight log but it happens quite often in SITL. (At least on our codebase)

dagar, 2018-11-20:
We can save it directly back to _params.mag_declination_source, then on the PX4 side the module needs to commit the parameter.

How about something like this? 
PX4/ecl: https://github.com/PX4/ecl/commit/779d4c41dda2a0f3cd35360f565b531186f8af40
PX4/Firmware: https://github.com/PX4/Firmware/commit/8e64584838c8ae501f5e1525f16a5c099dc4e975

CarlOlsson, 2018-11-21:
Looks like a nice cleanup!
I just scrolled through it quickly but could you please just explain how it solves the problem that _mag_declination_gps and _NED_origin_initialised is updated in collect_gps() but _mag_declination is only updated in calcMagDeclination() which doesn't run if _imu_updated is false?

CarlOlsson, 2018-11-21:
What do you think about changing
```_params.mag_declination_deg = math::degrees(_mag_declination);```
in save_mag_declination() to
```_params.mag_declination_deg = math::degrees(_mag_declination_gps);```

dagar, 2018-11-21:
Right, that race condition still exists. Yes your suggestion would do it.

CarlOlsson, 2018-11-27:
@dagar will you create a PR with your branch including my proposed change?

priseborough, 2021-08-17:
Current code sets the declination at the same point in code where the EKF origin is set. The function that accesses the declination checks that the origin has been set.

closed 2021-08-17 by priseborough