PX4/PX4-ECL #525: High susceptibility to noisy mag data

CarlOlsson, 2018-11-21:
At bootup we filter the mag measurements and then use the filtered value to initialize the earth magnetic field measurements, and if [this is merged](https://github.com/PX4/ecl/pull/524), also to initialize the heading.

However, as soon as we get GPS lock we again reinitialize the earth magnetic field states and heading, this time without filtering. The same is true also for the reinitializiation of the heading at every takeoff if EKF2_MAG_TYPE = 0 (automatic switching between heading and 3D axis mag fusion).

The obvious issue here is that two times every flight the raw value of the magnetometer is super important. There is no auto rejection of these samples, so if they are wrong the heading and mag states can be arbitrary far away from reality.

We of course also use the raw sensor measurement from e.g. GPS every time we reset the vel/pos states, but that should mostly happen if something is wrong, not every flight as is the case with the mag.

dagar, 2018-11-21:
This is probably crude, but the cost of running the mag filter continuously would be minimal. 

[image]

CarlOlsson, 2019-01-23:
For vehicles using EKF2_MAG_TYPE = 1 (only heading fusion) this is now even worse.
Since https://github.com/PX4/ecl/pull/544 we reinitialize the earth magnetic field states at takeoff and start using them for calculating the declination. If that single magnetometer sample that is used for this is bad the declination will be off and there is no way for the EKF to recover since the earth field states are not even estimated

PR #662 (2019-11-08): [ekf] controlMagFusion refactor and mag field strength check
I basically tried to abstract the overall mag fusion type switching from the implementation details and get rid of the multiple checks for the same flags (e.g.: `in_air`).

I still need to test a lot and maybe I'll move the whole controlMagFusion and sub-functions to a new file.

Feature added: the mag fusion is inhibited when the norm of the magnetic field exceeds X times the expected strength (from WMM table).

fixes #525
closed 2019-11-08 by bresch