PX4/PX4-ECL #528: Improve docs readme and confirm doc status

hamishwillee, 2018-11-29:
The documentation in https://github.com/PX4/ecl/tree/master/EKF/documentation was reported in [PX4 slack](https://px4.slack.com/archives/C0V533X4N/p1543389371534300) as requiring an update. I have posted request for further information.

FWIW I would have expected the README to have bullet point information about what each of the docs are for (Output Predictor.pdf, Process and Observation Models.pdf, Reference List.pdf), and I am wondering why there is no link to other EKF docs like https://dev.px4.io/en/tutorials/tuning_the_ecl_ekf.html

xdwgood, 2018-11-29:
> The Matlab script used to derive the autocoded expressions in the EKF can be found here: https://github.com/PX4/ecl/blob/master/matlab/scripts/Inertial%20Nav%20EKF/GenerateNavFilterEquations.m
A working Matlab model of the filter capable of replaying flight data can be found here: https://github.com/PX4/ecl/tree/master/matlab/EKF_replay

Need update:
https://github.com/PX4/ecl/blob/master/matlab/scripts/Inertial%20Nav%20EKF/GenerateNavFilterEquations.m  **----->** https://github.com/PX4/ecl/blob/master/EKF/matlab/scripts/Inertial%20Nav%20EKF/GenerateNavFilterEquations.m

https://github.com/PX4/ecl/tree/master/matlab/EKF_replay **---->** https://github.com/PX4/ecl/tree/master/EKF/matlab/EKF_replay

:smiley:  
FYI: @hamishwillee

xdwgood, 2018-12-06:
@hamishwillee  any update? :smiley:

hamishwillee, 2018-12-06:
@xdwgood No point asking me, this isn't under my control (hence posted this issue). @priseborough @CarlOlsson @mhkabir Do any of you own the above docs?

mhkabir, 2018-12-06:
@priseborough does, but the change is simple enough. It's just a link correction since the directories were rearranged slightly.

hamishwillee, 2018-12-07:
PR added. Do with it as you will. FYI @xdwgood nothing to stop you doing that too :-)

xdwgood, 2018-12-07:
I always feel that any suggestions on the PX4 are worth encouraging....But this time I am not happy.
Is the update of the document worthless?   It may also be the reason of the language, I can't understand what you mean.

hamishwillee, 2018-12-07:
@xdwgood Let me try again then.
- This was a good suggestion. Thank you.
- I have fixed the links in PR: https://github.com/PX4/ecl/pull/534

What I am saying is that there is no reason that you could not have submitted a PR to fix the links yourself. I'm very happy to help, but it doesn't seem like I added any real value given that you knew the problem and the solution.

xdwgood, 2018-12-07:
@hamishwillee  OK ，now I understand what you mean. Thank you

> What I am saying is that there is no reason that you could not have submitted a PR to fix the links yourself. 

Yes,I can indeed submit a PR myself, but I don't want to be an ecl contributor because of this unimportant change.  haha  ^_^  :smiley:

PR #534 (2018-12-10): Fix links to moved files in readme
Fixes #528
closed 2018-12-10 by priseborough