PX4/PX4-ECL #541: EKF: Problem with quaternion covariance reset at large tilt angles

priseborough, 2018-12-20:
When doing a yaw reset using the Ekf::initialiseQuatCovariances(Vector3f &rot_vec_var) method  (see https://github.com/PX4/ecl/blob/master/EKF/ekf_helper.cpp#L1429) if the vehicle has a large roll or pitch angle (eg tailsitter), the use of a rotation vector variance to represent the uncertainty does not enable the uncertainty in yaw to be changed without affecting the tilt.

A better method is required.

priseborough, 2018-12-20:
I've derived equations that should enable the yaw uncertainty to be increased. This will eliminate the need for a all of nothing reset of quaternion covariances following a reset of the yaw angle.

See  https://github.com/PX4/ecl/tree/pr-ekfYawCovReset

This will be implemented in the new year.

priseborough, 2018-12-23:
Further investigation of the code showed that this part of a wider issue that can affect all vehicle types that need to use an in-air reset to recover from a ground based magnetic yaw error. This is due to:

1) The preservation of off-diagonal elements in the covariance matrix in the 0-3 (quaternion state index) rows and columns that can lead to incorrect fusion of GPS and other NE frame observations following a state reset
2) The failure to preserve exisiting correlation information in the [4x4] quaternion state covariance sub-matrix when a full reset is applied.

Fundamentally the exisiting approach to handling the reset of covariances after a yaw reset was 'all or nothing'. 

The investigation also revealed that the reset of the yaw angle to enable recovery from ground based magnetic anomally is conditional on switching to 3-axis fusion and can therefore fail to execute depending on the setting of EKF2_MAG_TYPE and the flight profile. Changes to fix this hae been pushed to https://github.com/PX4/ecl/tree/pr-ekfYawCovReset. 

A PR will be submitted follong further testing. To assist with this a modification to the sitl_gazebo simulaton is being used that includes a crude model of a ground based magnetic anomally - see https://github.com/priseborough/sitl_gazebo/tree/earthFrameMagError

priseborough, 2018-12-24:
I was able to simulate a large tilt angle at the time fo the yaw reset by using the 'make posix_sitl gazebo_tailsitter' option, however this did require disabling the "Preflight Fail: Airspeed Sensor stuck" chekc in the commander and changing the  step size and real time update rate in tailsitter.world to 250 Hz, eg     <max_step_size>0.004</max_step_size>
<real_time_update_rate>250</real_time_update_rate>

Result with only critical yaw reset bug fix applied: https://logs.px4.io/plot_app?log=7e5f40a7-00d7-49c7-a0a6-8d422144a907

Innovations:
[image]

Quaternion Variances:
[image]

Test with all https://github.com/PX4/ecl/tree/pr-ekfYawCovReset changes: https://logs.px4.io/plot_app?log=69dd8220-570d-4897-be79-e2290eedcbd7

Innovations:
[image]

Quaternion Variances:
[image]

CarlOlsson, 2019-01-04:
@priseborough 
I tested this for a quad in sitl
[image]

As can be seen the variance for the q[0] and q[3] is increased which is correct.

> I was able to simulate a large tilt angle at the time of the yaw reset by using the 'make posix_sitl gazebo_tailsitter' option

Are you sure about this? As far as I know standard PX4 uses a convention for tailsitters such that the coordinate frame is similar to that of a quad in hover. I.e. in cruise the tailsitter flies along -z direction.

Without doing a hand held test with a real system I am not sure how to test the reset at high pitch angles in SITL. However, I pulled in the changes to our code base and this is result

[image]

This seems incorrect since the variance of all quaternion states increased with a similar amount

CarlOlsson, 2019-01-07:
@priseborough 
I tested this code by adding the entire 4x4 quaternion covariance matrix to logging and running
```
make px4_sitl gazebo
```
[Link to logfile](https://logs.px4.io/plot_app?log=4021dadc-e1aa-4c60-abc9-f07074037787)

I then converted the quaternion covariance matrix to an Euler covariance matrix following [these equations](https://stats.stackexchange.com/questions/119780/what-does-the-covariance-of-a-quaternion-mean)
(They are the same as the ones used [here](https://github.com/PX4/ecl/pull/543/files#diff-6ff089a846fa4bd452bf076e28a0bef7R994))

However, the result is not as expected, the blue and green lines overlap so the largest increase in variance is in the middle diagonal element (pitch).
[image]

priseborough, 2019-01-08:
However if you plot the quaternion state variances, the biggest increase is in q3 and less for q0 with q1  and q2 variance increase being negligible  which is consistent with an increase in yaw uncertainty for a vehicle that is close to zero roll and pitch.

CarlOlsson, 2019-01-08:
Yes, which indicates that the code is working as intended. I just need to figure out why the conversion to Euler angles does not give the expected result

TSC21, 2019-01-08:
@CarlOlsson I think it's related with the 321 and 312 variations that Paul spoke of and I think I know how to handle those on the covariance law. I will do an update later today.

CarlOlsson, 2019-01-08:
Yes me to, I just did a test with a quad. Hover for some time, increase heading noise, hover, decrease heading noise
[image]

The middle element in the covariance matrix is definitely yaw and not pitch

CarlOlsson, 2019-01-09:
Okey, so the confusion around the quaternion convention is sorted out in https://github.com/PX4/ecl/pull/543

What is left for this PR is to verify that the yaw uncertainty is increasing correctly for all attitudes, e.g. at pitch 90 degrees. 
In order to do that I calculated the Jacobian of the Euler angles to quaternions for both a 321 and 312 sequence.
Comparing the yaw variance on the logfile described [here](https://github.com/PX4/ecl/pull/543#issuecomment-452350609) gives the following result

[image]

This seems correct.

I then simulated a flight where the vehicle flew in a square (standard attitude, so body z facing down). In the middle of every leg I triggered `uncorrelateQuatStates()` and ```increaseQuatYawErrVariance()```

This is the result
[image]

As can be seen the variance from the 321 sequence seems correct. but for the 312 sequence something is strange, for the east and west direction no increase of uncertainty is observed

CarlOlsson, 2019-01-09:
For reference, here is the G matrices 

```
    #  Yaw using 321 Euler sequence
    G321[0][0] = 2*q4*(-2*q3**2 - 2*q4**2 + 1)/((2*q1*q4 + 2*q2*q3)**2 + (-2*q3**2 - 2*q4**2 + 1)**2)
    G321[0][1] = 2*q3*(-2*q3**2 - 2*q4**2 + 1)/((2*q1*q4 + 2*q2*q3)**2 + (-2*q3**2 - 2*q4**2 + 1)**2)
    G321[0][2] = 2*q2*(-2*q3**2 - 2*q4**2 + 1)/((2*q1*q4 + 2*q2*q3)**2 + (-2*q3**2 - 2*q4**2 + 1)**2) - 4*q3*(-2*q1*q4 - 2*q2*q3)/((2*q1*q4 + 2*q2*q3)**2 + (-2*q3**2 - 2*q4**2 + 1)**2)
    G321[0][3] = 2*q1*(-2*q3**2 - 2*q4**2 + 1)/((2*q1*q4 + 2*q2*q3)**2 + (-2*q3**2 - 2*q4**2 + 1)**2) - 4*q4*(-2*q1*q4 - 2*q2*q3)/((2*q1*q4 + 2*q2*q3)**2 + (-2*q3**2 - 2*q4**2 + 1)**2)
    
    # Yaw using 312 Euler sequence
    G312[0][0] = 2*q1*(-2.0*q1*q4 + 2.0*q2*q3)/((2.0*q1*q4 - 2.0*q2*q3)**2 + (q1**2 - q2**2 + q3**2 - q4**2)**2) + 2.0*q4*(q1**2 - q2**2 + q3**2 - q4**2)/((2.0*q1*q4 - 2.0*q2*q3)**2 + (q1**2 - q2**2 + q3**2 - q4**2)**2)
    G312[0][1] = -2*q2*(-2.0*q1*q4 + 2.0*q2*q3)/((2.0*q1*q4 - 2.0*q2*q3)**2 + (q1**2 - q2**2 + q3**2 - q4**2)**2) - 2.0*q3*(q1**2 - q2**2 + q3**2 - q4**2)/((2.0*q1*q4 - 2.0*q2*q3)**2 + (q1**2 - q2**2 + q3**2 - q4**2)**2)
    G312[0][2] = -2.0*q2*(q1**2 - q2**2 + q3**2 - q4**2)/((2.0*q1*q4 - 2.0*q2*q3)**2 + (q1**2 - q2**2 + q3**2 - q4**2)**2) + 2*q3*(-2.0*q1*q4 + 2.0*q2*q3)/((2.0*q1*q4 - 2.0*q2*q3)**2 + (q1**2 - q2**2 + q3**2 - q4**2)**2)
    G312[0][3] = 2.0*q1*(q1**2 - q2**2 + q3**2 - q4**2)/((2.0*q1*q4 - 2.0*q2*q3)**2 + (q1**2 - q2**2 + q3**2 - q4**2)**2) - 2*q4*(-2.0*q1*q4 + 2.0*q2*q3)/((2.0*q1*q4 - 2.0*q2*q3)**2 + (q1**2 - q2**2 + q3**2 - q4**2)**2)
```

TSC21, 2019-01-09:
@CarlOlsson have you tried to use the same G matrix for both 321 and 312 rotation orders? I tried to use the same for these and I seemed to get the same results on plots.

CarlOlsson, 2019-01-09:
> @CarlOlsson have you tried to use the same G matrix for both 321 and 312 rotation orders? I tried to use the same for these and I seemed to get the same results on plots.

I don't understand what you mean. The Jacobian (G matrix) is defined by the way you calculate the yaw angle which depends on the Euler sequence. How can you have different Euler sequences with the same Jacobian?

TSC21, 2019-01-09:
I can't really tell but for the 321 and 123 orders, applying the same G matrix, I get the same results. I can try to verify that on math.

CarlOlsson, 2019-01-09:
> We probably need to face that case as well, though I am not sure how we identify the rotation order on the function so to apply the proper G matrix.

You can probably use the same method as [here](https://github.com/PX4/ecl/blob/master/EKF/mag_fusion.cpp#L441)

For the logfile mentioned above, here is the raw quaternions
**East**
[image]

**North**
[image]

**West**
[image]

**South**
[image]

CarlOlsson, 2019-01-10:
I tested this code again, this time on a sitl quad.
I planned a square mission ([log](https://logs.px4.io/plot_app?log=1ed88263-d4ef-4901-80e5-9148a4e376f8)), all waypoints are 45 second loiter
[image]

Then at every waypoint I triggered uncorrelateQuatStates() and increaseQuatYawErrVariance() two times

This time I sometimes got 
```EKF mag yaw fusion numerical error - covariance reset```
when I triggered it.

Comparing the yaw variance calculated using the 321 and 312 Euler sequence looks like this
[image]

As can be seen they give similar result for the north/south direction but not for east/west. I don't know why at the moment.

yaw for the 312 calculation is given by
```yaw = atan2(-2.0 * (q2 * q3 - q1 * q4), q1 * q1 - q2 * q2 + q3 * q3 - q4 * q4)```

and the G matrix is:
```
G[0][0] = 2*q1*(-2.0*q1*q4 + 2.0*q2*q3)/((2.0*q1*q4 - 2.0*q2*q3)**2 + (q1**2 - q2**2 + q3**2 - q4**2)**2) + 2.0*q4*(q1**2 - q2**2 + q3**2 - q4**2)/((2.0*q1*q4 - 2.0*q2*q3)**2 + (q1**2 - q2**2 + q3**2 - q4**2)**2)
G[0][1] = -2*q2*(-2.0*q1*q4 + 2.0*q2*q3)/((2.0*q1*q4 - 2.0*q2*q3)**2 + (q1**2 - q2**2 + q3**2 - q4**2)**2) - 2.0*q3*(q1**2 - q2**2 + q3**2 - q4**2)/((2.0*q1*q4 - 2.0*q2*q3)**2 + (q1**2 - q2**2 + q3**2 - q4**2)**2)
G[0][2] = -2.0*q2*(q1**2 - q2**2 + q3**2 - q4**2)/((2.0*q1*q4 - 2.0*q2*q3)**2 + (q1**2 - q2**2 + q3**2 - q4**2)**2) + 2*q3*(-2.0*q1*q4 + 2.0*q2*q3)/((2.0*q1*q4 - 2.0*q2*q3)**2 + (q1**2 - q2**2 + q3**2 - q4**2)**2)
G[0][3] = 2.0*q1*(q1**2 - q2**2 + q3**2 - q4**2)/((2.0*q1*q4 - 2.0*q2*q3)**2 + (q1**2 - q2**2 + q3**2 - q4**2)**2) - 2*q4*(-2.0*q1*q4 + 2.0*q2*q3)/((2.0*q1*q4 - 2.0*q2*q3)**2 + (q1**2 - q2**2 + q3**2 - q4**2)**2)
```

priseborough, 2019-01-11:
Where are you getting the derivation for the conversion from  a quaternion rotation to a 312 Tait-Bryan sequence and the derivation for the associated partial derivative matrix?

CarlOlsson, 2019-01-11:
The conversion from quaternion rotation to a rotation matrix comes from [here](https://github.com/PX4/ecl/blob/master/EKF/ekf_helper.cpp#L1354), and then to 312 Tait-Bryan yaw comes from [here](https://github.com/PX4/ecl/blob/master/EKF/mag_fusion.cpp#L546)

The derivation for the partial derivative matrix comes from the following python script
```
from sympy import *
q1, q2, q3, q4 = symbols('q1 q2 q3 q4')

yaw = atan2(-2.0 * (q2 * q3 - q1 * q4), q1 * q1 - q2 * q2 + q3 * q3 - q4 * q4)
yaw_q1 = diff(yaw, q1)
yaw_q2 = diff(yaw, q2)
yaw_q3 = diff(yaw, q3)
yaw_q4 = diff(yaw, q4)

print('G[0][0] = ' + str(yaw_q1))
print('G[0][1] = ' + str(yaw_q2))
print('G[0][2] = ' + str(yaw_q3))
print('G[0][3] = ' + str(yaw_q4))
```

priseborough, 2019-01-14:
I've done further investigation today and the "EKF mag yaw fusion numerical error - covariance reset" could be the result of setting the off-diagonals to zero and having fusion take place before the next covariance prediction. How were you triggering uncorrelateQuatStates() and increaseQuatYawErrVariance() ? I will add code to the branch so that the fusion is blocked until after the next covariance prediction.

Using Matlab I independently derived the Tait-Bryan first rotation (yaw) in a 312 sequence as a function of the quaternions and got

yaw = atan2(2*q0*q3 - 2*q1*q2, q0^2 - q1^2 + q2^2 - q3^2);

d_yaw/d_q0 =

(((2*q3)/(q0^2 - q1^2 + q2^2 - q3^2) - (2*q0*(2*q0*q3 - 2*q1*q2))/(q0^2 - q1^2 + q2^2 - q3^2)^2)*(q0^2 - q1^2 + q2^2 - q3^2)^2)/((2*q0*q3 - 2*q1*q2)^2 + (q0^2 - q1^2 + q2^2 - q3^2)^2);

d_yaw/d_q1 =

-(((2*q2)/(q0^2 - q1^2 + q2^2 - q3^2) - (2*q1*(2*q0*q3 - 2*q1*q2))/(q0^2 - q1^2 + q2^2 - q3^2)^2)*(q0^2 - q1^2 + q2^2 - q3^2)^2)/((2*q0*q3 - 2*q1*q2)^2 + (q0^2 - q1^2 + q2^2 - q3^2)^2);

d_yaw/d_q2 =

-(((2*q1)/(q0^2 - q1^2 + q2^2 - q3^2) + (2*q2*(2*q0*q3 - 2*q1*q2))/(q0^2 - q1^2 + q2^2 - q3^2)^2)*(q0^2 - q1^2 + q2^2 - q3^2)^2)/((2*q0*q3 - 2*q1*q2)^2 + (q0^2 - q1^2 + q2^2 - q3^2)^2);

d_yaw/d_q3 =

(((2*q0)/(q0^2 - q1^2 + q2^2 - q3^2) + (2*q3*(2*q0*q3 - 2*q1*q2))/(q0^2 - q1^2 + q2^2 - q3^2)^2)*(q0^2 - q1^2 + q2^2 - q3^2)^2)/((2*q0*q3 - 2*q1*q2)^2 + (q0^2 - q1^2 + q2^2 - q3^2)^2);

[TaitBryan312.m.txt](https://github.com/PX4/ecl/files/2753786/TaitBryan312.m.txt)

priseborough, 2019-01-15:
OK, so I will focus the investigation for the time being on the cause of the 'EKF mag yaw fusion numerical error - covariance reset' events. One thing to try is to ensure the covariance prediction runs before the next fusion step.

I would like to be able to reproduce your results in my testing using the same mission plan and firmware modifications.

priseborough, 2019-01-16:
Thank you to @CarlOlsson and Sebastian for spotting a sign error that was introduced when the auto-code had been cleaned up by hand in the increaseQuatYawErrVariance function. The https://github.com/PX4/ecl/tree/pr-ekfYawCovReset branch has been rebased has been rebased and the error fixed here: 
https://github.com/PX4/ecl/commit/69c29ad0e5e6db60ebc29ac537e87fa5301c29ff

A compile error introduced when the a rebase conflict was resolved has also been fixed - see 
https://github.com/PX4/ecl/commit/47cf16f8e3f727370aab0e2524f35d31d95ff1a3

CarlOlsson, 2019-01-16:
@priseborough As discussed I pushed two commits to this branch.
3d716c438d28e0d0c2c42205a8bdbddccab146d7 Also fill in the lower part of the covariance matrix
d36be5b3bd3707c805fc6773906f2ba4dce87498 Limit the yaw variance increase to 0.01 rad^2

priseborough, 2019-01-17:
@CarlOlsson I've pushed two commits to the branch which should address the issue of unstable declination immediately following a field reset when using 3-axis fusion and the loss of declination covariance information when switching in and out of 3-axis fusion in auto mode in into 3-axis fusion with EKf2_MAG_TYPE = 0

CarlOlsson, 2019-01-18:
Actually not only [this](https://github.com/PX4/ecl/issues/554) problem seems to be initializing the covariances to zero

There is also a problem when we call fuseDeclination() in resetMagYaw(). When we call this function the first time we activate 3d fusion we also call fixCovarianceErrors(), and here we set the mag field state covariances to zero since _control_status.flags.mag_3D is false

priseborough, 2019-01-20:
The covariance reset to 0 bug has been fixed here: 
https://github.com/PX4/ecl/commit/ad9414e4dadc6a7a25d33b7edc219775d8a88086

SITL log: https://logs.px4.io/plot_app?log=39f319e4-847f-467c-9056-78fb917c0515

LorenzMeier, 2019-01-21:
@priseborough I've put up a PR for easier reference to that commit: https://github.com/PX4/ecl/pull/556

CarlOlsson, 2019-01-21:
@priseborough Some comments to the updated code in this branch

1) The `fuseDeclination()` function in `resetMagHeading()` is never triggered [here](https://github.com/PX4/ecl/blob/ad9414e4dadc6a7a25d33b7edc219775d8a88086/EKF/ekf_helper.cpp#L705-L705) because `_control_status.flags.mag_3D` is always false.

2) This is actually also on master but since https://github.com/PX4/ecl/pull/544 the reset logic in controlMagFusion() is never triggered
https://github.com/PX4/ecl/blob/ad9414e4dadc6a7a25d33b7edc219775d8a88086/EKF/control.cpp#L1455-L1470
since `_control_status.flags.mag_align_complete` is always set to true [here](https://github.com/PX4/ecl/blob/ad9414e4dadc6a7a25d33b7edc219775d8a88086/EKF/control.cpp#L1391-L1391)

priseborough, 2019-01-21:
@CarlOlsson  I will try another approach to simplify the way fuseDeclination() is only called, either immediately before fuseMag() if it hasn't been called since the last yaw reset or immediately after if _control_status.flags.mag_dec is true. This will simplify the overall logic.

Discussion continued in PR.

mhkabir, 2019-02-01:
Fixed on master.

closed 2019-02-01 by mhkabir