PX4/PX4-ECL #545: TECS throttle and pitch calculation have got some quirks in them

almaaro, 2019-01-01:
Hello, 
I made a pull request concerning this issue already, but I think that this topic would actually require more generic discussion before an actual PR (I closed the original PR). I'll explain some parts of throttle and pitch calculation in tecs.cpp (at least how I've understood it):

`// Calculate the demanded true airspeed`
`_update_speed_setpoint();`
- This calculates _TAS_rate_setpoint from the airspeed error and _speed_error_gain.

`// Calculate the demanded height`
`_update_height_setpoint(hgt_setpoint, baro_altitude);`
- This calculates _hgt_rate_setpoint from the altitude error and _height_error_gain and the demanded climb/descend rate and _height_setpoint_gain_ff
1. quirk: why is _height_setpoint_gain_ff even a parameter? Why it should be something else than 1? If it is not 1, it will always result in a constant altitude error during climbs/descends. 

`// Calculate the specific energy values required by the control loop`
`_update_energy_estimates();`
- This just calculates height, airspeed, height rate and airspeed rate setpoints to specific energies.

`// Calculate the throttle demand`
`_update_throttle_setpoint(throttle_cruise, rotMat);`
- First it calculates the predicted throttle from the STE_rate_setpoint.
2. quirk: 
`// Calculate gain scaler from specific energy error to throttle
float STE_to_throttle = 1.0f / (_throttle_time_constant * (_STE_rate_max - _STE_rate_min));`
- If I am getting it right, STE_to_throttle is just a proportional term to change STE_error to throttle output assuming a linear relationship between throttle and the STE rate. But this assumes that _STE_rate_max is achieved at 100% throttle, not throttle_max.

3. quirk    
    `// Add proportional and derivative control feedback to the predicted throttle and constrain to throttle limits
_throttle_setpoint = (_STE_error + _STE_rate_error * _throttle_damping_gain) * STE_to_throttle + throttle_predicted;`
- This is taking the _STE_error again into account in throttle calculation. _update_speed_setpoint() and _update_height_setpoint() already converted the airspeed and altitude errors into rate setpoints, which was already used for throttle calculation.

- After this the integrator term is added.
- My suggestion would be to replace STE_to_throttle with just 1.0f / _throttle_time_constant and to add the derivative and integrative terms (the _throttle_integ_state would need to be calculated before this) directly into STE_rate_setpoint like this:
`_throttle_integ_state = _throttle_integ_state + (_STE_error * _integrator_gain) * _dt / _throttle_time_constant;`

    `float STE_rate_setpoint = constrain((_SPE_rate_setpoint + _SKE_rate_setpoint + _STE_rate_error * _throttle_damping_gain / _throttle_time_constant + _throttle_integ_state), _STE_rate_min, _STE_rate_max);`

`// Calculate the pitch demand`
`_update_pitch_setpoint();`

4. quirk: The height/airspeed rate setpoints calculated in _update_speed_setpoint() and 
_update_height_setpoint() get multiplied by SPE_weighting and SKE_weighting. So now the user has 
to take the weighting into account when setting  _height_error_gain and _speed_error_gain 
parameters.

5. quirk: _SEB_error is already taken into account in SEB_rate_setpoint (apparently _pitch_time_constant controls how _SEB_error is transferred into SEB rate):
`float SEB_correction = _SEB_error + _SEB_rate_error * _pitch_damping_gain + SEB_rate_setpoint * _pitch_time_constant;`

- My solution has been to limit the weights to the maximum of 1 (to prevent making the gains too large and unstable) and to use the actual SPE rate derived from pitch angle to calculate the throttle demand (so _update_throttle_setpoint() would come after _update_pitch_setpoint()):
`// Calculate a specific energy correction that doesn't include the integrator contribution`
    `float SEB_correction = _SEB_rate_error * _pitch_damping_gain + SEB_rate_setpoint ;`

    `// Divide it into kinetic and potential proportions for throttle calculations. Also normalise it in relation to _pitch_time_constant`
    `float SEB_rate_setpoint_pitch = constrain(SEB_correction / climb_angle_to_SEB_rate, _pitch_setpoint_min, _pitch_setpoint_max) * (_tas_state * CONSTANTS_ONE_G);`
    `_SKE_rate_setpoint_pitch = - SEB_rate_setpoint_pitch * SKE_weighting / (SKE_weighting + SPE_weighting);`
    `_SPE_rate_setpoint_pitch = SEB_rate_setpoint_pitch + _SKE_rate_setpoint_pitch;`

- So STE_rate_setpoint would then be calculated like this (the _SKE_rate_setpoint would be unchanged):
`float STE_rate_setpoint = constrain((_SPE_rate_setpoint_pitch + _SKE_rate_setpoint + _STE_rate_error * _throttle_damping_gain / _throttle_time_constant + _throttle_integ_state), _STE_rate_min, _STE_rate_max);`

- So for example if FW_T_SPDWEIGHT is set to 0.5, it keeps the SPE rates unchanged but reduces the SKE rate pitch control by half.

Please let me know what you think! After some discussion, I can propose some fully tested PRs. Also, just letting you know that I've got two other ideas concerning 1. throttle calculation that takes the airspeed into account 2. throttle integrator enhancements.

Best regards, 
almaaro

RomanBapst, 2019-01-27:
@almaaro I did not go through all of the the details but I noticed that you might be confused by the design of the control loops of the version of TECS that PX4 uses.
Similar to you I was also once curious about the design and compared it to the "original" implementation of TECS. 
You can read about that design here:
https://patentimages.storage.googleapis.com/99/6a/bc/aed12b8b2d6fd2/US4536843.pdf

I then talked to @priseborough  who is the author of the specific version of TECS that is implemented in PX4 and asked him about the differences in design. In the end it boiled down to differences between large transport aircraft (for which the original TECS was derived) and small UAVs. I also think @priseborough has simulink models for TECS and he might share that with you if you ask him.

In an effort to better document TECS I created control diagrams for both the pitch and the throttle loop, see below. My next step is to write a tuning guide for TECS with some plots showing reference performance.

[image]

[image]

It would be great if we could discuss on the basis of these diagrams. Also if you are willing to contribute to a documentation effort that would be great. Feel free to ping me.

almaaro, 2019-04-13:
Hi,
I made another control diagram to visualize the pitch setpoint calculation from the altitude point of view (no SKE data included). As you can see, the pitch time constant and the SPE weighting interfere with the update_pitch_setpoint() method. 

Also, the throttle calculation uses the SKE and SPE proportional and rate errors as the starting point. However, the actual SPE rate (the one that is the result of the actual pitch setpoint) differs from these values because of the applied SPE weighting.
[image]

On solution to these problems is in the first post (at the 5th quirk).

priseborough, 2021-08-17:
See https://github.com/PX4/PX4-Autopilot/pull/16362 for changes made that improve the control structure for a small UAV application.

closed 2021-08-17 by priseborough

kaklik, 2022-05-07:
> In an effort to better document TECS I created control diagrams for both the pitch and the throttle loop, see below.

Hello!  Do you have an original editable source of that diagrams, please? I am asking because the controller diagrams [diverge from the TECS source code considerably](https://discuss.px4.io/t/fixed-wing-tecs-controller-disconnect-with-documentation/19983). Therefore someone should update the diagrams to match the current source code. Unfortunately, it is a lot of work in cases where only jpg or png bitmaps for the diagrams are available.