PX4/PX4-ECL #553: Include yaw uncertainty in covariance reset

CarlOlsson, 2019-01-10:
In the [initialiseCovariance()](https://github.com/PX4/ecl/blob/b920910483657a9680749651995ac8e119fb5120/EKF/covariance.cpp#L49-L49) function, which is called e.g. if the heading fusion has encountered a numerical error, we don't account for the yaw uncertainty when initializing the quaternion covariances. This may result in a smaller yaw uncertainty after the reset than before.

@priseborough Is this intended or should we add the yaw uncertainty to the quaternion covariance after we have initialized them using the initial tilt error parameter?

priseborough, 2019-01-13:
It was originally written for use at startup prior to fusion of any data and. The likelihood of it being called in-flight is small - I have not seen this triggered since the early days of filter development when we were still finding the correct combination of process noise and covariance matrix error correction methods.

However if the observation method being used for yaw is known at the time of reset, then it would be preferable to use the corresponding yaw uncertainty.

priseborough, 2019-04-04:
@CarlOlsson I think this has been fixed by recent PR's addressing yaw reset and covariance issues. Can you confirm. TXS.

CarlOlsson, 2019-04-04:
I didn't check very thoroughly but to me it seems like the issue is still there. If the magnetic fusion hits a numerical issue we trigger [resetMagCovariances()](https://github.com/wingtra/ecl_private/blob/1601a4523c1da491371c51495c21b81ff68dc71d/EKF/covariance.cpp#L885-L903) which removes the knowledge of the uncertainty in yaw. However, as you said it is very unlikely (I don't think I ever saw it happen in a real flight).
So feel free to close here

bresch, 2020-06-23:
We've never seen any issue related to this and as said, the likelihood of this occurring in flight is almost null. Closing.

closed 2020-06-23 by bresch