PX4/PX4-ECL #567: Use vision position using gyro for yaw

hdiethelm, 2019-02-01:
Hello,
I have a drone flying indoor using UWB for positioning. This works quite nice as long the compass is not drifting to much. Which is sadly mostly the case because of a bad compass and magnetic steel parts all around.
Because the flight time is short (3-4 min) I was thinking it should be possible to deactivate the compass and use just the gyro as reference for yaw.
So I set:
EKF2_AID_MASK to 24 (vision position / yaw)
EKF2_EVA_NOISE to 10 rad (Don't trust my yaw value sent over vision message at all which is sent as 0 to initialize yaw=0 at startup)
However, the estimated yaw value drifts quite a lot:
[image]
while the integrated gyro value is stable and the integrated Z gyro shows a drift of just 2.5 deg:
[image]

All values are in deg and deg/s. I did this tests while the drone was staying on my desk. At the beginning, I rotated it by 360 deg in 90 deg steps and back to test the gyro gain.

As a second try I commented out https://github.com/PX4/ecl/blob/1a1106878d6ab1cc48565bef9b25b07d626c5d0b/EKF/control.cpp#L324 to avoid heading fusion for yaw at all. However, the results are the same.

Is there a numerical problem or a bug? Did I miss some settings? Or is not avoidable by the way the Kalman filter works?

hdiethelm, 2019-02-01:
Thanks!
I just did an other test: By forcing the gyro bias to zero, it gets better. But still, it seams that there is a bug because the integral of the gyro values is much more stable than the estimation.
```diff
diff --git a/EKF/control.cpp b/EKF/control.cpp
index f704168..3caf138 100644
--- a/EKF/control.cpp
+++ b/EKF/control.cpp
@@ -321,7 +321,7 @@ void Ekf::controlExternalVisionFusion()
 
 		// determine if we should use the yaw observation
 		if (_control_status.flags.ev_yaw) {
-			fuseHeading();
+			//fuseHeading();
 
 		}
 
diff --git a/EKF/ekf_helper.cpp b/EKF/ekf_helper.cpp
index 4246556..5940def 100644
--- a/EKF/ekf_helper.cpp
+++ b/EKF/ekf_helper.cpp
@@ -793,7 +793,7 @@ void Ekf::constrainStates()
 	}
 
 	for (int i = 0; i < 3; i++) {
-		_state.gyro_bias(i) = math::constrain(_state.gyro_bias(i), -0.349066f * _dt_ekf_avg, 0.349066f * _dt_ekf_avg);
+		_state.gyro_bias(i) = 0;
 	}
 
 	for (int i = 0; i < 3; i++) {
@@ -1196,7 +1196,7 @@ void Ekf::fuse(float *K, float innovation)
 	}
 
 	for (unsigned i = 0; i < 3; i++) {
-		_state.gyro_bias(i) = _state.gyro_bias(i) - K[i + 10] * innovation;
+		_state.gyro_bias(i) = 0;
 	}
 
 	for (unsigned i = 0; i < 3; i++) {
```

[image]

[image]

Logs:
[logs.zip](https://github.com/PX4/ecl/files/2821436/logs.zip)

log_3_2019-2-1-11-40-30.ulg No fuseHeading
log_4_2019-2-1-11-59-38.ulg fuseHeading 10 rad noise
log_7_2019-2-1-12-41-10.ulg fuseHeading 10 rad noise / gyro bias = 0
log_9_2019-2-1-12-57-54.ulg No fuseHeading / gyro bias = 0

By the way: I'm using a pixracer temperature calibrated / PX4 stable

hdiethelm, 2019-02-01:
I did an other test,  fuseHeading active and 10 rad noise. The drone was just standing on my desk, receiving zero as position and angle. At 291 s, something got out of control without any visible reason and yaw started drifting away. Meight be it was not visible before, because at about this time I rotated the drone.

[image]
[image]

[log_11_2019-2-1-13-58-16.zip](https://github.com/PX4/ecl/files/2821668/log_11_2019-2-1-13-58-16.zip)

hdiethelm, 2019-02-01:
Last test for today: Same results on PX4 master branch:
[image]
[image]
[log_12_2019-2-1-14-38-40.zip](https://github.com/PX4/ecl/files/2821773/log_12_2019-2-1-14-38-40.zip)

mhkabir, 2019-02-01:
@priseborough can you please look into this.

priseborough, 2021-08-17:
We can fly without a magnetometer with the current EKF using EKF2_MAG_TYPE=5. Also the updated covariance prediction enables smaller values of gyro bias process noise to be used to take advantage of IMU's that have  small thermal drift.

closed 2021-08-17 by priseborough