PX4/PX4-ECL #571: EKF: tilt_align and uncertainty at high pitch angels

CarlOlsson, 2019-02-06:
We declare `tilt_align = true` when the roll + pitch uncertainty is lower than 3 degrees.
https://github.com/PX4/ecl/blob/c52f4bab1a1655f298778b9a815bbe3c70d91529/EKF/control.cpp#L61-L62

This does not work properly at high pitch angles in the same way as the increase in yaw uncertainty at a heading reset wasn't working before.

E.g. for a tailsitter which is booted up with the body x axis facing upwards the tilt_aligned flag is set to true when the pitch and **yaw** uncertainty is below 3 degrees.

Sidenote:
Currently yaw_align is set to true in [initialiseFilter()](https://github.com/PX4/ecl/blob/dd58e695494acab73fb6fcec7074291edb5c226b/EKF/ekf.cpp#L265-L265) before tilt_align is set to true in [controlFusionModes()](https://github.com/PX4/ecl/blob/dd58e695494acab73fb6fcec7074291edb5c226b/EKF/control.cpp#L62-L62).
This leads to the fact that the magnetometer is fused before tilt_align is true

priseborough, 2019-04-04:
@CarlOlsson I think this has been fixed by https://github.com/PX4/ecl/pull/549, https://github.com/PX4/ecl/pull/552 and https://github.com/PX4/ecl/pull/556

CarlOlsson, 2019-04-04:
No, the issue is still there.

This makes an assumption that the vehicles z axis is pointing down
https://github.com/PX4/ecl/blob/f95cd4b3584d029c35b288a39180ddf25b2dd004/EKF/control.cpp#L56-L59

priseborough, 2021-08-17:
We could use the component of the angle_err_var_vec vector perpendicular to the earth frame D axis to determine of the tilt has aligned.