PX4/PX4-ECL #575: 3D mag fusion is broken

CarlOlsson, 2019-02-27:
If `EKF2_MAG_TYPE = 2` (only 3D mag fusion) the estimator will continuously trigger yaw resets
https://github.com/PX4/ecl/blob/1378ec179717448404c74819871d149217350b19/EKF/control.cpp#L1542-L1545

since mag_align_complete is set to false here
https://github.com/PX4/ecl/blob/1378ec179717448404c74819871d149217350b19/EKF/control.cpp#L1378-L1378

To reproduce run
`make px4_sitl gazebo`, set `EKF2_MAG_TYPE = 2` and restart simulation

[image]
[image]

dagar, 2019-02-27:
Any idea when it was last functional?

CarlOlsson, 2019-02-27:
No, I never used it before. Just wanted to test something in SITL and noticed that the vehicle never got GPS lock

closed 2019-04-07 by priseborough