PX4/PX4-ECL #576: Docs: link broken

julianoes, 2019-02-28:
The link in the readme for "EKF Documentation and Tuning Guide" does not work anymore.
FYI @hamishwillee

julianoes, 2019-02-28:
It's still here:
https://github.com/PX4/px4-devguide/blob/master/en/tutorials/tuning_the_ecl_ekf.md
and here:
https://web.archive.org/web/20180619060008/https://dev.px4.io/en/tutorials/tuning_the_ecl_ekf.html

hamishwillee, 2019-03-01:
Hi @julianoes 

The doc moved into the user guide: https://docs.px4.io/en/advanced_config/tuning_the_ecl_ekf.html
I created a redirect from the devguide https://dev.px4.io/master/en/tutorials/tuning_the_ecl_ekf.html
Unfortunately the master build did not propagate into the root, so if you go to https://dev.px4.io/en/tutorials/tuning_the_ecl_ekf.html you don't get that redirect.

The source for the devguide is here: https://github.com/PX4/Devguide

I have no idea what https://github.com/PX4/px4-devguide - the link you provided is. Can we tell who created it? Can it be deleted. @dagar do you know about this one?

mrpollo, 2019-03-04:
Can we fix the link in the README to point to the new location? this will fix the README

dagar, 2019-03-06:
No idea where https://github.com/PX4/px4-devguide came from. Should we get rid of it?

hamishwillee, 2019-03-06:
> Should we get rid of it?

@dagar Yes please. I don't have the permissions.

dagar, 2019-03-06:
> > Should we get rid of it?
> 
> @dagar Yes please. I don't have the permissions.

Done

hamishwillee, 2019-03-06:
Thanks. We still need to fix issues with the root not being updated :-)

PR #579 (2019-03-18): README: Fix link to EKF/ECL tuning guide
Fixes #576
closed 2019-03-06 by hamishwillee