PX4/PX4-ECL #581: Unused parameters warning

fvantienen, 2019-03-05:
We now need to compile with `-Wno-unused-parameter` in order to not get any warnings due to the `estimator_interface.h` having functions with unused parameters. Wouldn't it be nicer to define the `__attribute__((unused))` for these parameters specifically?
This way we don't need to compile the whole autopilot project with the `unused-parameter` flag.

dagar, 2019-03-05:
Are these parameters completely unused in ecl or only in your particular use case?
Are you building with cmake?

fvantienen, 2019-03-05:
These following functions have unused parameters:
- `virtual bool collect_gps(uint64_t time_usec, struct gps_message *gps) { return true; }` (`estimator_interface.h`, all parameters)
- `bool Ekf::collect_gps(uint64_t time_usec, struct gps_message *gps)` (`gps_checks.cpp`, only `time_usec`)

I am not building with cmake, we use our own Makefile system.

dagar, 2019-03-05:
Let's just update them. I'll fix the PX4/Firmware usage.

fvantienen, 2019-03-05:
You can then also remove the `-Wno-unused-parameter` from the cmake file in this repo.

closed 2019-03-05 by dagar