PX4/PX4-ECL #582: `make test` fails to build

tstellanova, 2019-03-05:
While investigating a potential bug in DataValidator, I want to add a unit test. 
When I run `make test` I get the error below.  
Any suggestions on how to move past this? Thanks!

```
blip:ecl todd$ make test
-- The CXX compiler identification is AppleClang 10.0.0.10001145
-- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++
-- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- set build type to RelWithDebInfo
-- PX4 ECL: Very lightweight Estimation & Control Library v0.9.0-647-g1378ec1
-- adding EKF python tests
-- Found SWIG: /usr/local/bin/swig (found version "3.0.10") 
-- Found PythonLibs: /usr/local/Frameworks/Python.framework/Versions/3.7/lib/libpython3.7m.dylib (found suitable version "3.7.2", minimum required is "3") 
-- Found PythonInterp: /usr/local/bin/python3 (found suitable version "3.7.2", minimum required is "3") 
CMake Warning (dev) at /usr/local/Cellar/cmake/3.13.3/share/cmake/Modules/UseSWIG.cmake:564 (message):
  Policy CMP0078 is not set.  Run "cmake --help-policy CMP0078" for policy
  details.  Use the cmake_policy command to set the policy and suppress this
  warning.
Call Stack (most recent call first):
  EKF/swig/python/CMakeLists.txt:17 (swig_add_library)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Configuring done
-- Generating done
-- Build files have been written to: /Users/todd/Documents/proj/firm-px4/src/lib/ecl/build/test_build
[2/51] Performing download step (git clone) for 'matrix'
Cloning into 'matrix'...
Already on 'master'
Your branch is up-to-date with 'origin/master'.
[49/51] Swig compile ../ecl_EKF.i for python
FAILED: EKF/swig/python/CMakeFiles/_ecl_EKF.dir/ecl_EKFPYTHON_wrap.cxx EKF/swig/python/ecl_EKF.py 
cd /Users/todd/Documents/proj/firm-px4/src/lib/ecl/build/test_build/EKF/swig/python && /usr/local/Cellar/cmake/3.13.3/bin/cmake -E make_directory /Users/todd/Documents/proj/firm-px4/src/lib/ecl/build/test_build/EKF/swig/python /Users/todd/Documents/proj/firm-px4/src/lib/ecl/build/test_build/EKF/swig/python/CMakeFiles/_ecl_EKF.dir && /usr/local/Cellar/cmake/3.13.3/bin/cmake -E env SWIG_LIB=/usr/local/Cellar/swig/3.0.10/share/swig/3.0.10 /usr/local/bin/swig -python -outdir /Users/todd/Documents/proj/firm-px4/src/lib/ecl/build/test_build/EKF/swig/python -c++ -I/Users/todd/Documents/proj/firm-px4/src/lib/ecl/build/test_build/matrix-prefix/src/matrix -I/Users/todd/Documents/proj/firm-px4/src/lib/ecl -I/usr/local/Frameworks/Python.framework/Versions/3.7/include/python3.7m -I/Users/todd/Documents/proj/firm-px4/src/lib/ecl/EKF/swig/python/../.. -o /Users/todd/Documents/proj/firm-px4/src/lib/ecl/build/test_build/EKF/swig/python/CMakeFiles/_ecl_EKF.dir/ecl_EKFPYTHON_wrap.cxx /Users/todd/Documents/proj/firm-px4/src/lib/ecl/EKF/swig/ecl_EKF.i
/Users/todd/Documents/proj/firm-px4/src/lib/ecl/build/test_build/matrix-prefix/src/matrix/matrix/SquareMatrix.hpp:70: Error: Syntax error in input(3).
ninja: build stopped: subcommand failed.
make: *** [test_build] Error 1
```

dagar, 2019-03-05:
SWIG doesn't seem to like one of the recent PX4/Matrix changes (https://github.com/PX4/Matrix/commits/master). 

As a temporary workaround you could point it at https://github.com/PX4/Matrix/commit/6b0777d815cd64902eb0575d56ec52f53aebb4a0 until we find the proper fix. 
https://github.com/PX4/ecl/blob/1378ec179717448404c74819871d149217350b19/CMakeLists.txt#L125-L132

dagar, 2019-03-05:
Temporary workaround - https://github.com/PX4/ecl/pull/583

dagar, 2019-03-05:
Looking forward to more testing on this side. We have code coverage for standalone PX4/ecl tests - https://codecov.io/gh/PX4/ecl. This doesn't yet include testing when included in PX4/Firmware.

tstellanova, 2019-03-07:
Thanks for the pin workaround. `make test` builds now, and I was able to get further with testing, but now I’m thwarted by python test harness dependency problems.  It strikes me that having tests in CI for just this library is crucial to assuring it performs as intended. If I have time in the future I might tackle getting CI tests integrated— I think there was another feature request issue already that asked for this, and I think @dagar already commented on how to get started there. The main unknown for me is how to integrate with the CI framework that would be most straightforward for dronecode to operate for this library going forward. IIRC there are multiple CI frameworks in use.

closed 2019-03-07 by tstellanova

dagar, 2019-03-07:
I agree that we should have significantly more testing on this side. ECL either needs to exist as a true standalone project (consumed by multiple other projects) or might as well merge back into PX4/Firmware.