PX4/PX4-ECL #595: Replay conversion script outdated: baro and mag missing from sensor_combined

jbeyerstedt, 2019-04-08:
I tried to run [EKF/matlab/EKF_replay/Common/convert_px4_sensor_combined_csv_data.m ](https://github.com/PX4/ecl/blob/master/EKF/matlab/EKF_replay/Common/convert_px4_sensor_combined_csv_data.m), but this does not work any more.
```
Undefined function or variable 'baro_timestamp_relative'.
```

When taking a close look, all baro and mag data is missing from the imported CSV file.
baro_timestamp_relative.
If I identified it correctly, https://github.com/PX4/Firmware/commit/3b5b12e1d16b9c603aaa920f0f1ea96f6848f3db  (from 2019-02-18) introduced the changes, which break this script.

priseborough, 2019-06-03:
Yes, you would need to use an earlier version of firmware to generate a compatible log. Feel free to submit a PR that fixes https://github.com/PX4/ecl/blob/master/EKF/matlab/EKF_replay/Common/convert_px4_sensor_combined_csv_data.m

jbeyerstedt, 2019-06-13:
I have a patched the script for now, but I don't have the time now to submit a pull request.
Because I also created some additional scripts to automate the whole process by running a bash script calling Matlab from the command line. And ground truth data will also be imported.
But the bash script is quite specific to my use case for now and will also only find the Matlab executable for a single Matlab version and on macOS. So it's not ready to be released yet.

In about 8 weeks, when my thesis is done, I can take a look at this again.

PR #630 (2019-09-23): fix #595 and add new all-in-one conversion script (PX4 only)
fixes #595.
Because [EKF/matlab/EKF_replay/Common/convert_px4_sensor_combined_csv_data.m ](https://github.com/PX4/ecl/blob/master/EKF/matlab/EKF_replay/Common/convert_px4_sensor_combined_csv_data.m) will not use one timestamp and relative times any more, multiple timestamps are needed. So you need to be careful about the timestamps while importing the data from the CSV file.
Therefore I created a new MATLAB script, which imports all CSV files and converts them into the MATLAB data files.

Additionally you can convert ground truth dat [... cut]
commit fe3fec41 (2019-09-23): fix #595 and add new all-in-one conversion script (PX4 only)
closed 2019-09-23 by priseborough