PX4/PX4-ECL #602: Suboptimal wind estimation initialization

CarlOlsson, 2019-05-10:
[Here](https://logs.px4.io/plot_app?log=264d2d6d-026c-4ee0-a277-cd826c993ec6) is a sitl mission with gazebo plane where the vehicle takes of facing east and does a loiter circle
[image]
Synthetic sideslip fusion was disabled in the test

One would obviously expect the wind uncertainty in the east direction to be initialized with a lower uncertainty than in north direction. However this is not the case
[image]

Here is another example of airspeed fusion commencing after transition to cruise on a vtol platform 
[image]

As can be seen the uncertainty in east is in the beginning much higher than north.
I guess this has to do with the fact that the airspeed fusion measurement model does not incorporate the fact that the pitot tube measures the airspeed only in one direction, more info [here](https://github.com/PX4/ecl/issues/399).

We should add this as an alternative fusion method.

CarlOlsson, 2019-05-10:
An intermediate step would be to use the fact that we know that the pitot measures the airspeed in one direction for the covariance intialization in the same way as we use it in the wind state initialization
https://github.com/PX4/ecl/blob/a27a43eafa8f4dd514e89984f5394260a36ea4f6/EKF/airspeed_fusion.cpp#L267-L267

priseborough, 2019-05-13:
That is correct. The derivation assumes an omni directional sensor. I can derive an alternative observation model that uses a cosine law.

CarlOlsson, 2019-05-13:
okey awesome, I started with that some time ago but got stuck on fact that you have to take the partial derivate of the discontinues function.
If you know away around it that would be awesome!

priseborough, 2019-05-14:
The initialisation of wind states and covariances does take direction into account - see 

https://github.com/PX4/ecl/blob/master/EKF/covariance.cpp#L920-L921

and

https://github.com/PX4/ecl/blob/master/EKF/airspeed_fusion.cpp#L266

However, if the pitch angle is high, then the initialisation could be incorrect due to the use of yaw angle from a 321 rotation sequence.

priseborough, 2019-05-14:
The initialisation assumes a 10 deg sideslip uncertainty, however given yaw estimation errors, this should be increased

CarlOlsson, 2019-05-14:
@priseborough 
I derived an alternative wind state covariance initialization method as follows:
1) The initial wind state uncertainty in the body frame is `[R_TAS 0; 0 sq(initial_wind_uncertainty)]`, I.e. the airspeed sensor noise parameter defines the uncertainty along body X, and the initial_wind_uncertainty parameter defines the uncertainty in body Y
2) This covariance matrix is then as usual rotated to the intertial frame by pre and post multiplying with the jacobian of the rotation from body to intertial frame

The implementation is here and I tested it with a SITL mission where the plane takes off and fly straight east (sideslip fusion deactivated)

**master/PR**
[image]

As can be seen now the uncertainty of the wind state orthogonal to the takeoff direction (controller by _params.initial_wind_uncertainty) is initialized with a higher value than the the state parallel to the airspeed sensor (controller by _params.eas_noise)

CarlOlsson, 2019-05-14:
Update:
I added the uncertainty in yaw angle (for now also 10 degrees) resulting in a slight increase in state uncertainty
**without accounting for yaw uncertainty/with accounting for yaw uncertainty**
[image]

Code is here
https://github.com/PX4/ecl/blob/0db075448d903daf31f5315933085e05891c11ee/EKF/covariance.cpp#L912-L926

And derived using
```
import sympy as sp
Wx, Wy, yaw, R_TAS, initial_wind_uncertainty_var, R_yaw = symbols('Wx Wy yaw R_TAS initial_wind_uncertainty_var R_yaw')
Wn = Wx * sp.cos(yaw) - Wy * sp.sin(yaw)
We = Wx * sp.sin(yaw) + Wy * sp.cos(yaw)

Wn_Wx = sp.diff(Wn, Wx)
Wn_Wy = sp.diff(Wn, Wy)
Wn_yaw = sp.diff(Wn, yaw)
We_Wx = sp.diff(We, Wx)
We_Wy = sp.diff(We, Wy)
We_yaw = sp.diff(We, yaw)

G = sp.Matrix([[Wn_Wx, Wn_Wy, Wn_yaw],[We_Wx, We_Wy, We_yaw]])
wind_body_cov = sp.Matrix([[R_TAS, 0.0, 0.0], [0.0,initial_wind_uncertainty_var, 0.0], [0.0, 0.0, R_yaw]])
wind_cov = G * wind_body_cov * G.T

print('P[22][22] = ' + str(wind_cov[0,0]))
print('P[22][23] = ' + str(wind_cov[0,1]))
print('P[23][22] = ' + str(wind_cov[1,0]))
print('P[23][23] = ' + str(wind_cov[1,1]))
```

Antiheavy, 2019-05-14:
@CarlOlsson @priseborough 

Here is a fixed wing flight log we had that showed bad initial wind estimate.  Feel free to look at it if you think it could be useful in validating the improvement you are working on here.

https://review.px4.io/plot_app?log=8b9df1c8-82ca-417b-9f76-30fde4dd190a

priseborough, 2019-05-15:
Derivation looks OK. It would be worthwhile thinking of the body frame wind relative velocity as an airspeed and sideslip (ie polar) rather than a cartesian x,y in body frame as that will scale with different airspeeds for a given variation/uncertainty in sideslip.

priseborough, 2019-05-15:
It would probably be easier just to scale the y axis uncertainty using an assumed sideslip uncertainty and the expected value of airspeed.

CarlOlsson, 2019-05-15:
@Antiheavy 
In the logfile you linked above the wind states seems to be initialized to zero
[image]

I am not sure which software was running in the plane but this was a bug some time ago https://github.com/PX4/ecl/issues/399

CarlOlsson, 2019-05-15:
@priseborough I changed the initial wind uncertainty in body Y to be dependent on the initial uncertainty in sideslip angle
https://github.com/PX4/ecl/blob/0db075448d903daf31f5315933085e05891c11ee/EKF/covariance.cpp#L912-L928

I did not use the sideslip noise parameter since the expected sideslip might be significantly higher in the initialization, e.g. due to the fact that the vehicle just finished transitioning to cruise flight

Antiheavy, 2019-05-15:
> I am not sure which software was running in the plane but this was a bug some time ago #399

@CarlOlsson the firmware was based on PX4 v1.8.0 - so I might have not had that bug fix.  Thanks for letting me know.

closed 2021-08-17 by priseborough