PX4/PX4-ECL #604: Error in ekf_helper.cpp

ecpknpaul, 2019-05-21:
Hi:
ekf_helper.cpp:
line 512:
`// Note q1 *= q2 is equivalent to q1 = q2 * q1 `
I think it should be `q1 *= q2 is equivalent to q1 = q1 * q2`, because in file Quaternion.hpp, line 272-276:

> void operator*=(const Quaternion &other)
    {
        Quaternion &self = *this;
        self = self * other;
    }

and I think in file ekf_helper.cpp, line 519
`_output_new.quat_nominal = _state_reset_status.quat_change * _output_new.quat_nominal;`
should be:
`_output_new.quat_nominal =  _output_new.quat_nominal * _state_reset_status.quat_change;`

Am I right？Looking forward to your reply.

CarlOlsson, 2019-05-21:
Hi,

yes you are right! I forgot to fix those lines in [this PR](https://github.com/PX4/ecl/pull/586)
I did not notice since it was not visible in the logs since the error only affects one single output.
How did you find it? 
I pushed a fix [here](https://github.com/PX4/ecl/pull/605)

ecpknpaul, 2019-05-21:
Hi,
I just read the code,  and find the quaternion multiplication is in the wrong order.

priseborough, 2019-05-23:
Yes, the comment and code are inconsistent and we should be using the same convention for multiplication order as the Eigen library. Please provide log evidence of the error because if the order is wrong it should be visible in a log.

closed 2019-06-17 by priseborough