PX4/PX4-ECL #612: Yaw Airspeed Scaler Different from Pitch and Roll

supergra, 2019-05-31:
According to [this controller document](https://dev.px4.io/en/flight_stack/controller_diagrams.html), all three of the attitude controllers should apply quadratic IAS scaling to the P gain, and linear TAS scaling to the FF gain.

This is reflected in both `ecl_roll_controller.cpp` and `ecl_pitch_controller.cpp`:

```
/* Apply PI rate controller and store non-limited output */
	_last_output = _bodyrate_setpoint * _k_ff * ctl_data.scaler +
		       _rate_error * _k_p * ctl_data.scaler * ctl_data.scaler
+ _integrator; //scaler is proportional to 1/airspeed
```

But not in `ecl_yaw_controller.cpp`:
```
	/* Apply PI rate controller and store non-limited output */
	_last_output = (_bodyrate_setpoint * _k_ff + _rate_error * _k_p + _integrator) * ctl_data.scaler *
ctl_data.scaler; //scaler is proportional to 1/airspeed
```

Is this a bug?

bresch, 2019-06-11:
@supergra Yes, it is a mistake, I remember fixing that about a year ago on a private fork but I forgot to upstream the fix :|
And the second mistake is that we scale both feedforward and the controller with IAS (`ctl_data.scaler = 1/IAS`). That is tracked in https://github.com/PX4/Firmware/issues/11975 but I'm waiting for the new "airspeed module" to make the change.

supergra, 2020-06-24:
@bresch Is there any obstacle to upstreaming your fix for the first bug (the yaw being different)? We were considering making the change locally, but I'm curious if you are holding off because it might introduce unexpected behavior. 

The second bug is likely a minor impact for most low altitude work we do, so not as concerned about that one.

supergra, 2020-07-04:
Thanks @bresch, I will make a pull request. Suggest transferring this issue to Firmware, since `fw_att_control` was moved there. (I don't think I can do that myself.)

bresch, 2020-07-06:
@supergra thanks a lot! I'll move the issue and review your PR asap.

closed 2020-07-10 by bresch