PX4/PX4-ECL #625: GPS heading fusion does not block mag heading fusion

AndreasAntener, 2019-07-28:
From the parameter documentation I understand that with GPS heading enabled, and mag fusion set to automatic, the mag yaw should only be used until the GPS heading is available. But for that I think there is a condition missing similar to the one already in place for EV heading fusion in https://github.com/PX4/ecl/blob/master/EKF/control.cpp#L1377, otherwise the mag fusion is re-enabled right after GPS heading fusion disables it.

I'd appreciate if someone can clarify that this is the general idea behind it. Also, heading fusion never falls back to mag if GPS becomes unavailable?

AndreasAntener, 2019-07-29:
If we assume magnetometer values are not safe to use in certain areas we don't want it use both or fall back to mag.

What happens if GPS heading is not updated anymore for a long time? This case is not explicitly handled yet as far as I can see. It seems to just continue integrating the gyro.

AndreasAntener, 2019-07-31:
This is from a bench test, you can see the innovation spikes which have roughly the length of the difference between mag and GPS heading. With MAG_TYPE 0. I don't think that's the idea.

[image]

RomanBapst, 2019-08-03:
@AndreasAntener As far as I know gps heading was tested but I don't think the logic is complete (as you have correctly stated). Do you have a replay log available which I could use to test some changes?
I do have a setup in the office in case you don't have one available. During the replay I can stop gps data to make sure that it switches over between gps and mag correctly.

AndreasAntener, 2019-08-06:
No replay log yet. Trying to reproduce locally. Similar issue like Paul reported occurs as well: https://github.com/PX4/ecl/issues/627

I will push the small change I did, which makes it switch correctly as far as I can see, and it works well, but we many times we have a drift initially which won't come to rest.

AndreasAntener, 2019-08-08:
Issue reproduced on PX4 1.9.2: https://review.px4.io/plot_app?log=703082d0-c57f-40a8-bdd4-5194de2233f5

[image]

priseborough, 2019-09-17:
The decision to not fall back to mag was deliberate. In the case where loss of GPS is extended, heading will dead reckon using gyro, but horizontal velocity vector changes will make it observable and apply corrections. I need to check that we have protection against the growth in yaw uncertainty resulting in the covariance matrix becoming badly conditioned. There is an exisiting method to prevent this for the case where we operate without a yaw sensor, eg when EKF2_MAG_TYPE = 4, see https://github.com/PX4/ecl/pull/628. 

The same method can be used to protect against bad conditioning of the covariance matrix for extended loss of GPS yaw when GPs heading is enabled.

priseborough, 2019-09-17:
I've merged https://github.com/PX4/ecl/pull/629 becasue it fixes the logic bug, but have left the issue open until the yaw drift and covariance conditioning questions have been answered.

priseborough, 2020-06-27:
The code now checks for quaternion variance growth in the fuseHeading() function when yaw fusion is inhibited and if too large, fuses a zero innovation yaw observation.

closed 2020-06-27 by priseborough