PX4/PX4-ECL #627: EKF heading drifts when EKF2_MAG_TYPE = 4

priseborough, 2019-08-06:
The EKF2_MAG_TYPE = 5 parameter setting is supposed to enable operation without magnetometer in situations where there are high levels of magnetic interference and no observations in an earth frame of reference that require an absolute heading such as GPS. Examples where this option may be used is indoor using optical flow.

Testing with EKF2_AID_MASK = 0 (to disable GPs use) and EKF2_MAG_TYPE = 4 has showed that the heading drifts on the ground because when the vehicle is stationary, the last non-moving heading is not being used as a yaw observation as was intended in the original design. Also there are no yaw innovations at any time resulting in a steady growth in quaternion variance that may eventually destabilise the covariance matrix.

The following log can be replayed to illustrate this effect:

https://logs.px4.io/plot_app?log=ac2c156b-14f2-46d3-87d8-3100353e1823

AndreasAntener, 2019-08-06:
Paul, could the same happen with EKF2_MAG_TYPE 5? We sometimes see a drift as well (right from boot) but using aid mask 129. I'm trying to reproduce and provide a log.

AndreasAntener, 2019-08-08:
Local test with mag type 5, interestingly this now shows an oscillation. Not sure what the data drop is about. Also it calms down quickly after GPS heading is available this time. It's quite hard to reproduce exactly as it happened on the field, but I hope this at least shows the initial cause (still trying to get the exact effect we've had).

https://review.px4.io/plot_app?log=7eb93866-a59d-429b-80e3-e0cde7a3e03d

[image]

AndreasAntener, 2019-08-08:
Oh, what I forgot to mention is that the drift (or oscillation) is matched by the gyro bias. It doesn't follow the actual gyro drift. Not sure if that is part of the cause or just another symptom. At least, from what I can see is, that is has nothing to do with gyro drift.

closed 2019-10-08 by priseborough