PX4/PX4-ECL #646: Optical flow fusion failing for unknown reasons with PMW3901 sensor

tstellanova, 2019-09-23:
Flight log is here:

https://logs.px4.io/plot_app?log=a5c47c33-1a45-4d0f-9aaa-336cc5b69259

Using pmw3901 optical flow sensor and TFMini Plus range sensor with Pixhawk 4.  No GPS is attached. EKF2_AID_MASK is set to 3.   Operator is simply carrying the pixhawk at walking speed at about 1m above ground level in a repeated loop.

ECL is rejecting X and Y flow: `innovation_check_flags` bitmask alternates between 0, 2048, 3072
(bitmask:
# 10 - true if the X optical flow observation has been rejected
# 11 - true if the Y optical flow observation has been rejected
)

Any suggestions on why optical flow would be rejected in this case?  Ground range sensor appears to be reporting valid distance.

tstellanova, 2019-09-24:
To be clear, this is using a build derived from latest PX4 master, so should include eg f557cdf88c791ae4a3ddb796bbbcf679933365f9 from PX4/Firmware

dagar, 2019-09-24:
Check the sensor rotation?