PX4/PX4-ECL #651: EV data ignored, not sure exactly why

nbelanger99, 2019-10-04:
We'll be debugging a bit more tomorrow, but so far, ev data is fused for some time, but will be ignored after some time even when the vehicle is sitting still. The log files reveal 
`EV HGT Timeout`
and later
`External Vision Data Stopped`

Looking at code: 
`bool reset_to_baro = !ev_data_available && baro_data_available;`
Is evidently what's failing. 

Does this imply a sensor latency issue? Why does it take some time (10-20 seconds or so) for this to occur? A buffer issue? 

Attached is a log file converted to txt 

[20_54_36.txt](https://github.com/PX4/ecl/files/3688744/20_54_36.txt)

kamilritz, 2020-01-15:
Is this issue still persisting? There was quite some work done on external vision fusion in the last months. I would recommend to try again with current master.

kamilritz, 2020-01-31:
The 'External Vision Data Stopped‘ error only appears if the EKF is stop receiving any new data. So probably the vision system stopped sending data. Also make sure that you send ev_data at a rate around 30 Hz.

closed 2020-01-31 by kamilritz