PX4/PX4-ECL #659: 【Question】Do we need a stricter gps check?

xdwgood, 2019-10-28:
issue： https://github.com/PX4/Firmware/issues/13279
Currently we have completed the gps check in this function:`
bool Ekf::gps_is_good(const gps_message &gps)`  

We use eph, epv to check the position drift. There is a problem here. The eph and epv conditions that we set by default are easy to satisfy.

In the https://github.com/PX4/Firmware/issues/13279 problem: the latitude and longitude data drift provided by the gps module causes the ekf to estimate the local position drift, but in fact the autopilot is not moved.
[image]

For me, I might be able to adjust the process noise of gps and reduce the trust in gps data in the position estimate.
But for a customer, if he takes off directly in this state, the drone position will drift，right?

I wonder if we should restrict eph, epv more strictly, or add a position drift detection when the drone is still on the ground.

Any reply or criticism is welcome

mhkabir, 2019-10-28:
Already implemented. See https://github.com/PX4/Firmware/issues/9549.

xdwgood, 2019-10-29:
I am already using the latest master branch, but local.x/y is still drifting when the autopilot is still on the ground.There are also no warnings preventing me from entering position mode.

RomanBapst, 2019-10-29:
@xdwgood It's up to you if you want to have stricter checks on the gps data.
Have a look at his table [here](https://docs.px4.io/master/en/advanced_config/tuning_the_ecl_ekf.html#gps-performance-requirements)
It describes the different parameters you can use to achieve your goals. You can actually specify a maximum drift rate.

closed 2019-10-29 by RomanBapst