PX4/PX4-ECL #682: Velocity and position test ratios are not logged correctly on PX4 v1.10-rc

CarlOlsson, 2019-12-12:
Here are velocity and position test ratios from PX4 v1.10 release candidate (from a random logfile I found on review.px4.io).
The upper is the release candidate and the lower is a logfile collected with an older version of ECL for comparison
[image]

As can be seen the upper is useless. Running a SITL mission with max logging frequency of the estimator_status on the release candidate shows spikes in the test ratios at 5 HZ
(again RC vs comparison)
[image]

Plotting the test ratio and the innovations shows that the test ratio is now reset to zero right after the fusion (upper) while in the past it was not (lower)
[image]

I will be out for a couple of weeks starting tomorrow so don't think I will have much time to fix this until the 5th of Jan.

CarlOlsson, 2019-12-12:
@kamilritz Seems like the issue comes from here https://github.com/PX4/ecl/pull/642/commits/e9f6b7140a0e56ad426dc0449d4c9c8aa8a5bbd5

What was the purpose of this change?

dagar, 2019-12-12:
This should be fixed for v1.10 shortly.

dagar, 2019-12-12:
Thanks for testing Carl!

kamilritz, 2019-12-12:
@CarlOlsson The purpose was to have the test_ratio reporting zero after the fusion of a particular sensor is stopped completely. 
You are right `e9f6b71` is causing this. When the `fuseVelPosHeight()` function is called with only `_fuse_height = true` it sets the `pos_test_ratio` and `vel_test_ratio` accidentally to zero.
This is a clear bug. 
I created a branch with a fix for the release. @dagar is getting it into the release.

@CarlOlsson Thank you very much catching this!

FYI @priseborough

CarlOlsson, 2019-12-12:
> @CarlOlsson Thank you very much catching this!

No problem :) We started to work on a rebase on v1.10 so will do a lot of testing

dagar, 2019-12-12:
Fixed in https://github.com/PX4/Firmware/pull/13736.

closed 2019-12-12 by dagar