PX4/PX4-ECL #744: EKF initialization is unreliable if the vehicle is moving at bootup

CarlOlsson, 2020-02-10:
The initial tilt and yaw alignment is based on the average value of the IMU and magnetometer measurements for 2 times the observation buffer, so normally 320 ms. 
If during this time the vehicle is moving (e.g. the user carries it) the initial attitude estimate will be bad resulting in diverging IMU biases, heading fusion innovation checks failing etc.

here is a plot from a replay logfile where I booted up the vehicle and then lifted it up and carried it around. Top plot is EKF initialization delayed until I already started moving and bottom one is original so EKF initializes when stationary.
[image]

One simple solution would be to wait with initialization until the [vehicle at rest flag](https://github.com/PX4/ecl/pull/743) is true. However, this could cause issues for users flying from moving platforms, e.g. boats. Another solution would be to greatly lower the time the EKF averages the sensor data for the tilt alignment. This obviously reduces the robustness against noise data.
@priseborough What do you think?

dagar, 2020-02-11:
I'm wondering about the history of the "2 times the observation buffer" requirement, if there's anything we can do (or have already done) with the actual drivers to ensure only good data is published initially.

CarlOlsson, 2020-02-11:
> I'm wondering about the history of the "2 times the observation buffer" requirement, if there's anything we can do (or have already done) with the actual drivers to ensure only good data is published initially.

We have added spike detection on baro and mag by comparing every new sample to the low pass filtered value. If the derivative is too large, reject the measurement
[image]

Didn't push it since it is such a hack but it is definitely needed to stop bad data as early as possible

dagar, 2020-02-11:
> We have added spike detection on baro and mag by comparing every new sample to the low pass filtered value. If the derivative is too large, reject the measurement
> [image]
>
> Didn't push it since it is [...]

Which mag is this and can you point to the version of the driver upstream?

CarlOlsson, 2020-02-12:
> Which mag is this and can you point to the version of the driver upstream?

hmc5883, PX4 v1.7.3, will let you know if it still happens on v1.10.1

CarlOlsson, 2020-02-24:
> hmc5883, PX4 v1.7.3, will let you know if it still happens on v1.10.1

It does
[image]

bresch, 2020-03-03:
@CarlOlsson I'm interested in a suitable solution for the "takeoff from moving platform" case. What do you think about checking the length of the specific force vector (like in a simple AHRS) instead of averaging everything?

CarlOlsson, 2020-03-03:
To me that sounds like a good idea. Since the IMU data is already downsampled before reaching the EKF I think averaging it again doesn't really add much value

priseborough, 2020-03-05:
See https://github.com/PX4/ecl/pull/768

closed 2020-03-05 by priseborough