PX4/PX4-ECL #771: Vehicle crash due do estimator GPS failure

ndepal, 2020-03-09:
We just had another vehicle crash. EKF2 reported a GPS glitch and a delta velocity bias estimate issue.

The firmware is based on PX4 v1.10.0.
We are using the Ublox F9P GPS module in the vehicle and on the ground.

The vehicle took off in mission mode and briefly after started drifting when it crashed into a tree.
At the point of takeoff, the vehicle had been turned on for ~6 hours.

The symptoms are very similar to the crash mentioned in https://github.com/PX4/Firmware/pull/14312#issuecomment-595762171, but this time we do have a log, fortunately.

Looking at the log, the `vehicle_gps_position` looks sane, whereas the `vehicle_global_position` and the local position does not. Therefore, I feel that the issue is not with the GPS measurements (i.e. they should not have been rejected) but rather with the estimator that falsely rejected the measurements.

As this is not the first incident of this kind, I would really appreciate some assistance in looking into this and preventing it from happening in the future.

The flight log can be found here: https://logs.px4.io/plot_app?log=c231b7b3-ab35-4f48-be8c-0cb0a836cae4

CarlOlsson, 2020-03-09:
Without more context and the preflight log it is hard to say for sure. But you definitely have magnetic interference on the ground disturbing the estimated heading.
[image]

I would first try to remove this. Second thing you can try is to update ECL to master since there is now a yaw recovery estimator running in the background, specifically designed to handle failure cases like this
https://github.com/PX4/ecl/pull/766

LorenzMeier, 2020-03-09:
@ndepal Given your mag interference issues I will close this issue since this is not a PX4 performance issue at the core - it is a specific use case (lots of metal around the vehicle) that requires a new feature (as below):

https://github.com/PX4/ecl/pull/766

ndepal, 2020-03-09:
Thanks @CarlOlsson for having a look and pointing out the magnetic interference. That is potentially an issue.

However, EKF2 already has a feature that resets the heading to the current mag measurement once the vehicle reaches a certain altitude (1.5m above ground) https://github.com/PX4/ecl/blob/dec1ed0aeb395a885b60b1d61cfb57bd932c7d34/EKF/control.cpp#L1451
(this code looks quite different in master, the above links to what PX4 v1.10.0 uses)
This reset happens at time 23100 (4.5s into the log), as seen by the `estimator_status/control_mode_flags` and by the fact that the heading innovations become small at that time. So, assuming that in air we have reasonable mag measurements (and I do believe that, we normally have good position hold performance), the on ground-mag issues should be a thing of the past at the time of GPS rejection.

Additionally, in the previous crash we flew for several minutes (with good position hold) and crashed only as we were flying back to land. For that crash, https://github.com/PX4/ecl/pull/766 would not have helped, as that is only used in the first 30s of flight.

So I'm not convinced that a) mag interference is to be blamed for the crashes, and b) the new failure recovery feature would have prevented the crashes.

CarlOlsson, 2020-03-09:
> So, assuming that in air we have reasonable mag measurements (and I do believe that, we normally have good position hold performance), the on ground-mag issues should be a thing of the past at the time of GPS rejection.

Yes but magnetic interference does not only affect the heading estimate. Again, without knowing more of the background or at least having the preflight logfile it is hard to say, but it looks like your delta velocity bias states have diverged before takeoff.

[image]

The reason for this can be the magnetic interference, or e.g. a high temperature gradient of the IMU (since the IMU is not temperature calibrated and it looks like you are using a standard v4 board).

This most likely leads to the bad velocity estimation after takeoff and the innovation checks failing triggering the emergency landing
[image]

ndepal, 2020-03-09:
We just had another crash with this drone. Not sure if related at all. This time the altitude estimate was the problem. The setpoint and estimated altitude was 4m above home, but baro and GPS measurements, as well as onboard video, show that the vehicle descended back to the ground. 5 seconds after hitting the ground, EKF2 says `[ecl/EKF] EKF baro hgt timeout - reset to baro`.

https://logs.px4.io/plot_app?log=fa9c406a-e45f-4327-92fe-209db667d7bf

ndepal, 2020-03-09:
> or e.g. a high temperature gradient of the IMU (since the IMU is not temperature calibrated and it looks like you are using a standard v4 board).

We are using a v4 based board (identical schematics, different board form factor).
I don't see any drastic temperature changes in the log.

ndepal, 2020-03-09:
Basically, what we're seeing is that whenever the estimator fails, the GPS/baro measurements would have been fine. So I would like to limit EKF2's ability to "mess things up" as much as possible.

ndepal, 2020-03-09:
In both crashes, `estimator_status/filter_fault_flags` reports mag numerical errors (even before takeoff). The only place where this can happen without an accompanying print statement is https://github.com/PX4/ecl/blob/dec1ed0aeb395a885b60b1d61cfb57bd932c7d34/EKF/mag_fusion.cpp#L773
which points to an unhealthy covariance matrix.

I do see that the logged covariance values are strange in that log. E.g. `estimator_status/covariance.06` completely collapses early on in the log of https://github.com/PX4/ecl/issues/771#issuecomment-596659524.

If I'm not mistaken, such covariance issues should not come up even if the measurements are bad. Rather, it points to numerical issues in the estimator.

jkflying, 2020-03-09:
@ndepal if you can live without bias estimations, you can disable them in the EKF_AID_MASK. But we should really figure out what is happening here. If you can get us a log-from-boot with SDLOG_PROFILE having replay enabled, and showing the issue, it makes it a ton easier to debug.

We've seen intermittent issues popping up from the delta-velocity bias estimation for a long time now, so I really do think this is a real problem and not something specific to your setup.

ndepal, 2020-03-09:
Thanks @jkflying 
Getting a log since boot could be tricky since it's not very reproducible (except for today...) and since our system is on for a long time that log would be huge. I'll see what we can do.

ndepal, 2020-03-09:
Looking at what happens when a covariance issue is detected, the columns and rows of the affecting states are zeroed out. But it looks like the diagonal element is set to zero as well.

https://github.com/PX4/ecl/blob/dec1ed0aeb395a885b60b1d61cfb57bd932c7d34/EKF/mag_fusion.cpp#L766

Is that how it's supposed to be? Should it not be reset to some reasonably large value?

ndepal, 2020-03-09:
> @ndepal if you can live without bias estimations, you can disable them in the EKF_AID_MASK.

We're going to try that. How do I decide that I "can live without bias estimations"?
I guess I'd prefer consistency over slightly improved performance (if/when biases are estimated correctly), but I'm worried about GPS/baro measurements being rejected because of a real IMU bias that I'm now no longer estimating.

CarlOlsson, 2020-03-10:
> We just had another crash with this drone. Not sure if related at all. This time the altitude estimate was the problem.

This logfile has extremely high magnetic interference
[image]

> I don't see any drastic temperature changes in the log.

I meant before the logfile starts but after the vehicle was powered. 

> I do see that the logged covariance values are strange in that log. E.g. estimator_status/covariance.06 completely collapses early on in the log of #771 (comment).

That is very interesting! I have never seen that before, I guess it is related to the very long (<20h, I guess static) preflight time. 

> Getting a log since boot could be tricky since it's not very reproducible (except for today...) and since our system is on for a long time that log would be huge. I'll see what we can do.

A normal logfile with a low logging rate would be better than nothing, then at least we can see when and how the covariances collaps.

> Looking at what happens when a covariance issue is detected, the columns and rows of the affected states are zeroed out. But it looks like the diagonal element is set to zero as well.

That also looks like a bug to me. FYI @priseborough

RomanBapst, 2020-03-10:
@ndepal I will have a look today in the afternoon. I think these cases a quite interesting because not many people do this kind of testing.

ndepal, 2020-03-10:
> This logfile has extremely high magnetic interference

It is important to note that from 75704 on, the drone is skidding along the ground. The crazy parts of the magnetic field norm are after the impact.

[image]

Here is a good log of the same vehicle in the same environment:

[image]

We certainly do have a visible change in magnetic field between on-ground and in-air of about 0.1 G. I'm not sure if that is big enough to be worrying.

> That is very interesting! I have never seen that before, I guess it is related to the very long (<20h, I guess static) preflight time.

Very likely. We moved to v1.10.0 in December. Before that, we were based on v1.7.4 for about 2 years. With that version we regularly flew systems that had been turned on for weeks, even months. We had a few incidents that were related to the estimator, but we seem to be having a lot more problems since updating.

CarlOlsson, 2020-03-10:
> It is important to note that from 75704 on, the drone is skidding along the ground. The crazy parts of the magnetic field norm are after the impact.

(Y)

> We certainly do have a visible change in magnetic field between on-ground and in-air of about 0.1 G. I'm not sure if that is big enough to be worrying.

As long as the vehicle is not moving around in the field with the disturbances I think it should be fine. But we have seen problems where users boot up the vehicle e.g. inside a car and then moves out and places it on the ground. Even if the disturbances are gone at takeoff they have already confused the delta vel biases etc.

> Very likely. We moved to v1.10.0 in December. Before that, we were based on v1.7.4 for about 2 years.

We are actually in exactly the same situation. Have not experienced any problems yet but will make sure to do some testing with long time before takeoff

dusan19, 2020-03-10:
I did the following experiment in HITL
- I run automated repeated missions:
    1) Takeoff and fly to a specified location
    2) Go back to the mission start location and land (precision landing)
    3) Wait 3 seconds
    4) Repeat 1)

- I run this until the drone gets an estimator failure and does an emergency land
- The flight controller is never rebooted
- The flight controller was on for a long time, but the simulator was started just before this experiment.
- I have the logs of all the flights
- There are a few logged print statements added to indicate when the mission is done or updated and similar stuff.

Results of the experiment:

1) I got the first estimator failure after less then 20 flights:
https://review.px4.io/plot_app?log=75b8bf4a-495a-4621-abaf-2989e8cd57c6
Critical navigation failure was reported and some innovation checks failure. This is not necessarily the same problem as the one in this issue. There is no numerical problems reported and covariances are not clearly weird.

2) I started running repeated missions again after previous fallback landing. After 17 flights the estimator fails again:
https://review.px4.io/plot_app?log=fe2e060b-9eed-4981-a7dd-6bf01b694a68
Here, additionally, a gps glitch was reported

3) The drone landed far away from home after the previous estimator failure, and was landed there for another 20 minutes when it started repeatedly reporting estimator failure, numerical errors, height innovation failures (which looks like the same problems as in the second log reported in this issue: https://github.com/PX4/ecl/issues/771#issuecomment-596659524)
https://review.px4.io/plot_app?log=d35b2d34-02ae-4b0a-9e24-1c6932f1a551
I dont have a log of this 20 minutes period between the flights, i only started logging once I realized its crashing. The log is of the drone landed and disarmed.

dusan19, 2020-03-10:
I made a very big log from boot and did repeated missions, and a twice between the missions left the drone landed for ~20min.
Twice during flights I got critical navigation failure with innovation_check_flags = 1.
Interestingly i got filter_fault_flags =  8 (the fusion of the magnetic heading has encountered a numerical error) while the drone was landed for longer time (~10 min), this error repeated frequently for more than a minute, and finally stopped. During this time innovation_check_flags switched between 0, 4 and 5

Log:
https://review.px4.io/plot_app?log=e2d7e8f0-d58a-43b0-b8a7-e25dcd124705

dusan19, 2020-03-11:
In the log from my previous comment, it can be seen that while the drone is landed, z velocity and position covariances are very slowly decreasing, until the moment when they start decreasing drastically and quickly collapse to values that are almost zero. When this happens nothing is being fused. 
This motivated me to make another HITL simulation log, where the drone is only landed the whole time:
The log is more than an hour long, and these numerical errors happen multiple times in batches.
https://review.px4.io/plot_app?log=b39db461-2927-4bc1-9b0b-e5310ecac16c
The covariance collapsing looks like the same problem as @ndepal had in the logs he posted when he got numerical errors.

dusan19, 2020-03-11:
the z velocity variance slowly decreases. In the predict covariance step, after some time z velocity predicted variance P[6] becomes negative so fixCovarianceErrors constrains it to 0. Since 0 < KHP[6][6], measurements stop being fused which causes the drone to drift away, and also _fault_status reports numerical error.

dusan19, 2020-03-11:
The variance decreases but i think it should converge to some value in this experiment. Instead it keeps decreasing to 0. There are these functions that are used to deal with numerical problems such as having negative variances, but at that point its too late already. The variances are bad all the time while they are smaller than the value they should have converged to.

priseborough, 2020-03-12:
I think its due to badly conditioned delta velocity bias estimates and covariances forming during the static period on ground. 

These are delta velocity bias states rescaled to equivalent m/s/s:
[image]

These are the corresponding variances:
[image]

Can you please confirm the git commit hash for the version of ecl you are using.

priseborough, 2020-03-12:
Something else worth noting is a time slip present when the logging starts , eg:

[image]

Time slip is calculated as  

_integrated_time_us += imu_dt;
_last_time_slip_us = (now - _start_time_us) - _integrated_time_us;

This indicates that the ekf2_main update is running slow and/or the IMU driver is not calculating the delta time correctly or dropping data.

If this slip is occurring as series of small 4msec steps, then it would indicate the ekf2 thread not keeping up.

priseborough, 2020-03-12:
The filter_fault_flag bit position 15 is going high, so the EKF was aware of the bad delta velocity bias estimates:

[image]

priseborough, 2020-03-12:
@ndepal Can you please provide a git commit hash for the version of ecl that was running for this flight.

priseborough, 2020-03-12:
@dusan19 Can you please provide a link to a compatible PX4/Firmware version for replay of your logs and also a lnk to the version of ecl you are using. Both are required for us to replay your logs using the elf2 replay facility, determine affected versions and devise a fix.

priseborough, 2020-03-12:
@dusan19 I did a quick replay check on your log using upstream master and it has to many dropouts to be used for that purpose.

ndepal, 2020-03-12:
@priseborough Thank you for your input.

We are based on PX4 v1.10.0, which uses ECL version https://github.com/PX4/ecl/tree/dec1ed0aeb395a885b60b1d61cfb57bd932c7d34. Our branch of ECL has two commits on top that implement getter functions for some internal data, but do not change the functionality of the code.

This is valid for both mine and @dusan19 s logs.

dusan19, 2020-03-12:
I just want to add that I repeated the experiment with the delta velocity bias estimation turned off, and got the same results.

CarlOlsson, 2020-03-12:
I collected a ~7 hour stationary replay logfile but did not see any covariance stability issues. However, the shape is a bit strange..
[image]
[image]

TSC21, 2020-03-12:
@CarlOlsson @ndepal can https://github.com/PX4/ecl/pull/601 by any chance influence what you are seeing happening on your logs?

ndepal, 2020-03-12:
@CarlOlsson was this in SITL or on hardware?

Thanks for pointing that out @TSC21 , sounds like it could be related. @RomanBapst what do you think?

CarlOlsson, 2020-03-13:
> @CarlOlsson @ndepal can #601 by any chance influence what you are seeing happening on your logs?

No I don't think so. Here is the same logfile replayed without and with Kahan summation
[image]

> was this in SITL or on hardware?

Real HW

Looked through the logfiles again and wow, the delta velocity bias do diverge
[image]

TSC21, 2020-03-13:
> > @CarlOlsson @ndepal can #601 by any chance influence what you are seeing happening on your logs?
> 
> No I don't think so. Here is the same logfile replayed without and with Kahan summation
> [image]

Ok. Bu I still believe there needs to be some regression introduced somewhere causing this. As @jkflying is saying, this is something that happened on other systems, and I believe it might be causing what is being seen in https://github.com/PX4/ecl/issues/777.

CarlOlsson, 2020-03-13:
> Ok. But I still believe there needs to be some regression introduced somewhere causing this. As @jkflying is saying, this is something that happened on other systems, and I believe it might be causing what is being seen in #777.

Maybe I wasn't clear above, the diverging bias states are from the replayed logfile, so there is definitely an issue that is easily reproducible. Just saying that removing the Kahan summation does not seem to solve it

TSC21, 2020-03-13:
> Maybe I wasn't clear above, the diverging bias states are from the replayed logfile, so there is definitely an issue that is easily reproducible. Just saying that removing the Kahan summation does not seem to solve it

@CarlOlsson that was clear yes. I am just saying that or this problem is there since ever or something introduced a regression and that needs to be tracked. If it isn't the Kahan summation then it's something else. I just went through the different commits before the PX4 1.10 release that might have influenced the way that the delta velocity bias gets updated.

priseborough, 2020-03-17:
This log has a high CPU load which could be related to the time slip present from start of log

[image]

The GPS is at 10Hz which drives down state variances compared a standard 5Hz setup.

Other than these items I'm out of ideas until we can replicate with logging enabled from boot.

ndepal, 2020-03-17:
We are working on getting you a log from boot until estimator failure.

> The GPS is at 10Hz which drives down state variances compared a standard 5Hz setup.

Are there recommended changes to be made to EKF2 parameters when using an F9 module, for which 10Hz is standard?

ndepal, 2020-03-27:
We had another crash yesterday, apparently for the same reason.
Almost the same setup as before, except that this vehicle has a different mag.
Bias estimation was still enabled.
So this is just another datapoint of the same situation.
https://logs.px4.io/plot_app?log=d484793a-778b-47fb-84d2-fd8a78fbbebe

dusan19, 2020-03-31:
Here is the log from boot until the estimator fault. The drone was disarmed the whole time.
https://drive.google.com/file/d/1iGelWoNPyiMEFAISZ-t4uf0l8_U5YOug/view?usp=sharing
The errors are not exactly the same as the ones we had in flight which caused the crash (@ndepal 's logs), but might be related, and also probably shouldn't happen in general.

ndepal, 2020-04-01:
Today we are reproducibly seeing this issue, even right after boot.

The vehicle is landed in its base station and not moving.
The problem occurs soon after rebooting the flight controller. We have rebooted several times and it keeps happening.

This is a log from boot where the estimator has the first filter fault (mag numerical error) after 227s:
https://logs.px4.io/plot_app?log=135821dc-7839-4eef-a6aa-ec875604ce8e

Some observations:
- Same firmware as in previously posted logs. Based on v1.10.0. EKF IMU bias estimation is inhibited.
- The vehicle is landed and is not being moved during this log. The magnetic field measurements are constant. Thus, problems arising from magnetic disturbance is unlikely.
- The GPS measurements are as constant as they can be. During the entire log, the GPS is in RTK fixed mode.
- The reported EPH is low at 0.014. EKF2 limits this to 0.5 before fusing. The GPS measurements come in at 10Hz.
- The CPU load is about 85%. We investigated why we have a higher CPU load than the standard X frame and noticed that dshot uses about 7.5% more than PWM. In addition, we some I2C devices connected, namely an irlock camera and in this log the beacon is visible and measurements are being published.
- `estimator_status.timeslip` increases several times in the log, although it's not obviously correlated to the fault occurrences. @priseborough if timeslip due to high CPU load is to blame here, what is acceptable timeslip? Do most setups have zero timeslip?

ndepal, 2020-04-01:
As this is currently very reproducible, I tried switching back to `EKF2_AID_MASK 1`, i.e. allowing bias estimation. With that setting I have not yet gotten a filter fault today, while with `EKF2_AID_MASK 5` (inhibit IMU bias estimation), it happens within minutes as in the log above.

dusan19, 2020-04-06:
@priseborough @RomanBapst 
I have done some debugging by replaying the latest @ndepal 's log, and have found some problems/bugs regarding the estimator stability that I want to point out in general but which also seem to be causing the failures described in this issue. Additionally I added a small fix which seems to fix or at least mask the problems in this issue.

Estimator reports numerical errors when fusing mag measurement, i.e. the P[6][6] z-velocity variance would become negative after updating Pnew=P-KHP. This seems to **not** be specific to z velocity variance, but that one is just the first to fail, when the variance was forced to stay high another variance collapsed.

Mag fusion actually doesnt change the z velocity variance, same hold for gps fusion. Only the height fusion shows to change P[6][6], and it reduces it. All of this behavior is as expected.
On the other hand covariance prediction should result in increasing the variances P[i][i] depending on the process noise and the model (F matrix).
Ppred = FPF' + Q. In this step variances should be increased by the minimal amount defined by the relevant elements of the Q matrix. Also diagonal elements of the FPF' have to be positive.
Sadly this is not always the case. When observing P[6][6] i see that the it starts decreasing at one point, and from that moment both prediction and fusion steps reduce P[6][6] and quite quickly it collapses to 0 and numerical error is reported.
F matrix is not explicitly generated in the estimator but rather some (probably autogenerated) equations are used to predict the variance. This equations consist of P[i][i] terms, k*P[i][j] terms and terms that do not contain P[i][j] which are associated with Q. 
Q terms are always positive while the rest, k*P[i][j] (not including the first term P[i][i]) is quite often negative, which means that a value is subtracted from the previous variance and the variance is decreasing.
The problem is actually much worse than it seems, because adding Q terms most of the time masks the problem since in total change in variance (kP[i][j]+Q) is positive. But the variance does not rise enough, it rises less than the minimal theoretical amount defined by Q. Once kP[i][j] becomes more negative than Q is positive, total predicted variance even decreases.

This seems to me as a serious stability problem of the estimator.

I applied a fix: when predicting the covariance matrix, i check if Pnew[i][i] is at least relevant Q term higher than Pprevious, if not I make it that much higher. This basically means that kP[i][j] terms for the computation of the variances are positive. Or rather that FPF' > P on the diagonal elements.

With this fix I can not reproduce the problem anymore. Also P[6][6] variance converges to a certain value.

If the variances decrease in the prediction step due to some precision, float, or other numerical problems. then this fix is reasonable.
Equally if not more likely the bug might be anywhere in the actual equations used in the prediction or fusion steps, and in that case this fix masks the problem and is not the right approach.

Any feedback regarding the correctness of my analysis and the fix are much appreciated.

priseborough, 2020-04-07:
The covariance prediction is using single precision operations on terms spanning several orders of magnitude. In the past we have increased the delta velocity process noise to mask this.  The log from boot provided will be very useful when testing a more permanent fix.

jkflying, 2020-04-07:
Hi @dusan19 could you post a PR with your fix, with the replayed log, and some screenshots showing the relevant different parts of the log you noticed before vs after the fix? I suspect when the value gets small enough we have some numerical issues which causes the variance to collapse. 

We might be able to add Kahan summation to the relevant sections if we can see exactly where it is coming from. Alternatively, having lower bounds on certain terms (in this case, making sure the value we're adding to the variance is positive) is a tool we've reached for to fix numerical issues as well, even if it isn't as ideal.

dusan19, 2020-04-07:
The fix I made is the last commit here https://github.com/dusan19/ecl/tree/cov_prediction_fix
It is based on ecl version that firmware 1.10 points to.
Although it does fix the P[6][6] collapsing problem that I was originally looking into, the problem shifts to other states which I will describe here.

Original replayed log shows that after a few minutes covariance P[6][6] collapses and it doesnt recover from this state for a long time:
[image]

With the fix the situation is as follows:
[image]
Here P[6][6] doesnt collapse, but P[9][9] and interestingly P[0][0] do collapse. They recover quickly, but still a failure is issued which doesnt solve the initial problem (invalid position estimation). During the time just before collapsing, the covariance is the following:
[image]
During the time before collapsing, the forceful variance increment kicks in, and it increases it for the minimal amount, which is not enough. The problem are the off-diagonal elements being wrong. Take for instance P[6][9], its negative, which means that z velocity is oppositely coupled with z position which is clearly wrong. This value is normally positive when the estimator doesnt crash. This value is also used when computing the variances after fusion, so probably the variance is reduced more than it should be due to having this wrong values that contradict the model. The problems with the off diagonal elements might come from numerical precision issues since there are no limitations such as ones imposed on the variances with this fix, and they should also be handled, but they might also be due to some other bugs.

Some other covariance related observations I found suspicious:
Initial covariance matrix is:
[image]
After some time (while the estimator works fine) the covariance matrix is:
[image]
We can see that some of the quaternion elements change sign compared to the initial state. I find that suspicious, it seems that we expect the states to be coupled oppositely of what P converges to.

Do you have an idea on how to tackle this off-diagonal elements, or what might be the issue?

priseborough, 2020-04-08:
Replay of the https://logs.px4.io/plot_app?log=135821dc-7839-4eef-a6aa-ec875604ce8e log shows that either turning bias estimation back on or applying a lower limit of 1E-8 to the velocity variances in the fixCovarianceErrors() function preventing the behaviour. I am investigating further.

priseborough, 2020-04-08:
@dusan19 I would expect the quaternion off diagonals to change from the initial post alignment values given the propagation of body frame delta angle process noise errors.

priseborough, 2020-04-08:
As a check that we haven't inadvertently modified the covariance prediction auto-code (which is shared with ArduPilot EKF3 which hasn't yet encountered this behaviour), I copied it across again from the source file with no change to behaviour.

I then looked at the the delta time being used by the covariance prediction, as small values for that are problematic due to the use of dt^2 in a number of expressions. The first check was to change FILTER_UPDATE_PERIOD_MS (which controls the average dt value achieved) from 8 to 12 msec, which was what we used in earlier ecl/EKF versions, and this eliminated the behaviour. 

I then reverted it back to 8 and printed the value of dt used in the covariance prediction. This showed that it is regularly dropping to 4.8msec. I then hardcoded the value of dt used to FILTER_UPDATE_PERIOD_S but that did not solve the issue indicating that it is the average value is more important. 

I then changed FILTER_UPDATE_PERIOD_MS to 10 and the vertical velocity variance collapse is postponed to right at the end of the log, eg:

[image]

Using FILTER_UPDATE_PERIOD_MS = 10 and hardcoding the dt used by the covariance prediction to 
FILTER_UPDATE_PERIOD_S totally eliminated the behaviour.

This testing suggests that reductions in FILTER_UPDATE_PERIOD_MS and jitter in the dt used in the covariance prediction are contributors. A multi-pronged approach is recommended which could include:

- Limiting the minimum value of the position and velocity variances to 1E-6
- Using the average value of dt, eg FILTER_UPDATE_PERIOD_S in the covariance prediction
- Increasing FILTER_UPDATE_PERIOD_MS from 8 to 10msec 

I will now check the inputs to the covariance prediction and fusion steps to make sure that we don't have any incorrect input values or fusion operations using deactivated states present.

Other items to check in slow time include:

Do the revised covariance prediction autocode generated by https://github.com/PX4/ecl/pull/762 makes a difference. Sometimes a different ordering of operations for the what is mathematically the same cam be better conditioned when numerical precision issues become noticeable.
Does use of  double precision operations make a difference? If not then maybe the derivation could be updated to include higher some higher order terms. If it does, then a more expensive method of updating the covariance (eg computing both upper and lower diagonals and averaging) could be used.

priseborough, 2020-04-08:
Performing the covariance prediction using both upper and lower diagonals and then averaging to force symmetry made no difference

priseborough, 2020-04-08:
The other check we to do is replay this using current ecl master as there were changes made to the covariance matrix manipulation since the version used to generate this log.

priseborough, 2020-04-08:
I'm unable to replay this log using current ecl master so this becomes a blocker to submitting a PR. @ndepal and @dusan19 are you able to do a ground test using upstream PX4/Firmware master and provide the log?

priseborough, 2020-04-08:
Latest news is that substituting the the revised autocode generated by https://github.com/PX4/ecl/pull/762 also solves the issue on replay with no other changes

priseborough, 2020-04-08:
See https://github.com/priseborough/ecl/commit/d1288279a87355e43357f38a0f3ef6db8869c5bc

Let's discuss the way forward in terms of which of the changes in  https://github.com/priseborough/ecl/tree/covariance-wip should be submitted as a PR against master and how we verify them.

In summary, the following changes were found to be effective:

Increasing FILTER_UPDATE_PERIOD_MS to 10msec and changing the covariance prediction to use an average dt of FILTER_UPDATE_PERIOD_S
Limiting the lower value of velocity variances to 1E-8 (would advise 1E-6 is the smallest value that makes sense with real world errors)
Replacing the covariance prediction auto-code with code generated using https://github.com/PX4/ecl/pull/762

priseborough, 2020-04-08:
Note, that 

https://github.com/PX4/ecl/blob/108e9762e024d305b1c91bb6b1fa3caf79c38b20/EKF/python/ekf_derivation/main.py#L21 

and 

https://github.com/PX4/ecl/blob/108e9762e024d305b1c91bb6b1fa3caf79c38b20/EKF/python/ekf_derivation/code_gen.py#L43

have to be modified to output nextP[i][j] and P[i][j] syntax for compatibility with the stable release version

ndepal, 2020-04-08:
> I'm unable to replay this log using current ecl master so this becomes a blocker to submitting a PR. @ndepal and @dusan19 are you able to do a ground test using upstream PX4/Firmware master and provide the log?

We could create a stationary log with upstream. However, I would not expect to be able to reproduce this issue easily, given how rarely it usually occurs and on upstream we would not have our own configuration and apps running (which is important if CPU load is still suspected to have an influence).

ndepal, 2020-04-08:
@priseborough are you also observing off-diagonal covariance entry sign reversals in your testing and are they affected by your modifications?

Would it be feasible to check or even enforce the validity of the covariance matrix based on off-diagonal entry signs? This could be done in a similar fashion as the filter fault checks like here https://github.com/PX4/ecl/blob/0623b2b02d2be1df00209aedf71b7f4f0afb70d2/EKF/mag_fusion.cpp#L749
We know the correct sign for at least the position/velocity state pairs. For bias states I guess the signs could change (?) and for the quaternion I have no intuition.

jkflying, 2020-04-08:
@priseborough  I really like the idea of switching to #762 for code generation - it's possible that the more aggressive expression simplification solves some numerical issues by cancelling terms, so we might avoid this entire issue.

Also I don't see an issue with having a hard lower limit on the vehicle position accuracy of 1mm - that's just adding some prior knowledge about our positioning systems - although I think it would be better to increase the process noise to achieve this rather than have a hard clamp. The process noise should actually be adjusted to be something more like ~3cm with a good RTK setup left for a long time. But hard clamp as a safety mechanism is still good.

@ndepal About the negative off-diagonals, I don't have intuition here, sorry. This sounds like a promising way of detecting numerical issues though.

dusan19, 2020-04-08:
> Replay of the https://logs.px4.io/plot_app?log=135821dc-7839-4eef-a6aa-ec875604ce8e log shows either turning bias estimation back on or applying a lower limit of 1E-8 to the velocity variances in the fixCovarianceErrors() function preventing the behaviour. I am investigating further.

I found that too (for the accel bias) in this log but we got numerical errors reported in cases when we had accel bias estimation active (never this quickly after boot), unfortunately we dont have logs from boot of those.

dusan19, 2020-04-08:
Do all of your results use the variance increment lower limit that i wrote or not? Just so I can bring my code to the same state as in your tests.

> applying a lower limit of 1E-8 to the velocity variances in the fixCovarianceErrors() function preventing the behaviour

When I put a lower limit to velocity variances in fixCovarianceErrors() to 1e-8, I still get the same errors, here are the plots with and without my variance increment limit "fix":
[image]

[image]

[24 more replies]

closed 2021-07-22 by ndepal