PX4/PX4-ECL #777: EKF2 vertical velocity estimate not consistent with change in vertical position during forward flight

ryankierteal, 2020-03-11:
I have been seeing some strange behavior with the EKF2 vertical velocity and position estimation, and I'm having trouble tracking down the root cause of the issue.

The behavior is an uncommanded increase in altitude during forward flight in position mode.  The estimator is being quite slow to track the changes in the vertical position.  Additionally the estimate of vertical velocity is actually indicating that the drone is going down while both BARO and GPS observations are reporting that the drone is increasing in altitude.
Here is log showing this behavior:
[https://review.px4.io/plot_app?log=441ba346-5f27-4883-8797-4637a8a7f3e0](https://review.px4.io/plot_app?log=441ba346-5f27-4883-8797-4637a8a7f3e0)

The drone is running code forked from v1.10.0
BMI088 ACCEL ODR 400 Hz, BW 40 Hz

priseborough, 2020-03-12:
I would recommend you modify the internal EKF update step defined by FILTER_UPDATE_PERIOD_MS to be an integer multiple of the 5msec, ie change it from 8 to 10

ryankierteal, 2020-03-12:
@priseborough 
> I would recommend you modify the internal EKF update step defined by FILTER_UPDATE_PERIOD_MS to be an integer multiple of the 5msec, ie change it from 8 to 10
Here's a log with FILTER_UPDATE_PERIOD=10.   In this case the BMP388 BARO data is at 22 Hz, and EKF2_AID_MASK = 5:
[https://review.px4.io/plot_app?log=71ace796-7611-4dfe-af0d-0f2f34ba214a](https://review.px4.io/plot_app?log=71ace796-7611-4dfe-af0d-0f2f34ba214a)

priseborough, 2020-04-30:
Looks like clipping of the Z accel - evidenced by both vertical velocity and position innovations diverging in a positive direction.
[image]

Also these correlate to high forward speed and pitching moment due to blade flap - see how front/rear ESC demands go low/high to counteract pitching moment
[image]

[image]

closed 2020-04-30 by priseborough

ryankierteal, 2020-04-30:
Thanks for taking the time to do a detailed analysis @priseborough!