PX4/PX4-ECL #784: ‘isfinite’ was not declared in this scope

ashirsat, 2020-03-20:
I am getting an error isfinite() was not declared in this scope in ekf_helper.cpp when I am trying to make px4 SITL for ros-kinetic with gazebo 7. 

px4 tag is v1.8.0

Does anyone have a solution to this?

bresch, 2020-03-20:
you can use `ISFINITE` https://github.com/PX4/ecl/blob/362a2dfa084b476fc52d040290e1ac013f847eea/ecl.h#L100

bresch, 2020-03-20:
Does it work on v1.10 or master?

ashirsat, 2020-03-20:
Nope. I cannot build v1.10 on ubuntu 16.04 running ros-kinetic. That's why i had to use v1.8.0. @mhkabir : Do you have any suggestion?

TSC21, 2020-03-20:
This is duplicated in https://github.com/PX4/Firmware/issues/14440. We should be using `PX4_ISFINITE()`, since its definition will depend on the target arch.

LorenzMeier, 2020-03-20:
We're not supporting Gazebo 7 any more, neither do we support Ubuntu 16.04. I'm closing this accordingly. PX4 is an open source project working on state-of-the-art flight controls, that also means we can't maintain support for a four year old Linux distribution.

closed 2020-03-20 by LorenzMeier

ashirsat, 2020-03-21:
Thanks @TSC21 for the info.

@LorenzMeier : I agree. Currently using the kinetic branch due to dependency on intel realsense for ros-melodic

TSC21, 2020-03-21:
@ashirsat the issue was solved already in #785.

ashirsat, 2020-03-21:
@TSC21 Yeah I just pulled and build succeeded. Thanks.