PX4/PX4-ECL #800: GPS/VIO yaw aiding without mag not possible

bresch, 2020-04-21:
From: https://discuss.px4.io/t/possible-to-initialize-ekf-without-mag-sensor/16115

- [ ] EKF2 needs magnetometer data to initialize, even if `EKF2_MAG_TYPE` is set to none (https://github.com/PX4/ecl/blob/2fa43419d2bd7319024682b311f52c78f9a86f52/EKF/ekf.cpp#L181-L184)
- [ ] When not using mag (TYPE == none), we artificially fuse zero innovations to constrain the variance growth and help with gyro learning. If we're fusing GPS/VIO yaw data, this isn't required https://github.com/PX4/ecl/blob/47624a0f021e2c187cc4763ad829afe2aa4fb11a/EKF/mag_control.cpp#L49-L56

mhkabir, 2020-04-21:
@dagar I remember I brought this up a few months back, and we collected it in a Google doc. Do you have a link to that doc?

dagar, 2020-04-29:
> @dagar I remember I brought this up a few months back, and we collected it in a Google doc. Do you have a link to that doc?

Shared on slack.

priseborough, 2021-08-19:
These issues appear to be addressed in upstream master at 8aecc64

closed 2021-08-19 by priseborough