PX4/PX4-ECL #832: High vibration -> insane altitude -> shoot-up in altitude mode. Could it be prevented?

MaEtUgR, 2020-05-29:
I flew a QAV250 and crashed because of unrelated problems. When I tried to fly with it again after the crash the vibration levels were quite a bit higher for invisible mechanical reasons but it still flew ok-ish in stabilized. When I switched to Altitude I got a heart attack because it shut up into the air with full thrust. Apparently the altitude estimate diverged and that might be expected given the sensor measurements but my question would be could it have been automatically prevented since the estimate went insane looking at the log but I could still "use" it to switch to altitude mode.

[image]

https://logs.px4.io/plot_app?log=79083ce8-35ca-4f2f-9e32-0c1addc1da3c

FYI @bresch

dagar, 2020-05-29:
We could do something if the accel is experience continuous clipping.

[image]

The vibration metric might be another idea.

[image]

dagar, 2020-05-29:
Can we flag vertical position/velocity validity?

[image]
[image]

garlinplus, 2022-02-09:
[image]
From above picture,it can be found that your imu az is error compared with az of gps vd derivative.

bresch, 2022-02-09:
Since the update on the covariance prediction derivation, I didn't see such divergence due to high vibrations and moderate clipping anymore. Let's close this issue and re-open one if we encounter the problem again.

closed 2022-02-09 by bresch