PX4/PX4-ECL #834: EKF2 z divergence upon takeoff

francelico, 2020-05-29:
_Note: this issue was produced flying PX4 v1.10.1_

** Describe the bug **
One of our craft s EKF reported it was sharply descending while it was taking off, causing the drone to climb to 3m high by itself, with no input from the pilot. It then became controllable again.

We are using EKF2_HGT_MODE to BARO with a range finder enabled, so that the drone uses the range finder data until a certain height and speed. 

So I am not sure if the issue was caused by **high vibrations in the accelerometer** (as #832 ) or if the **range finder data was not fused in the ekf** for the first couple seconds upon takeoff.

The flight was indoors, so using only the baro and accel could lead to inacurrate z estimation. What I suspect is that not all the range aid conditions were met:
1) Vehicle is in-air: Since the issue occured upon take off could it be that this remains true for too long? What is the reasoning behing not fusing range finder data when the vehicle is on the ground anyway>
2) Range data is valid: There was no errors reported from range finder performance counters pre and post flight so I don't think this is it.
3) Vehicle is no further than max_hagl_for_range_aid away from the ground / 4) Ground speed is not higher than max_vel_for_range_aid -> not this either, we set those parameters very high.
**5) Terrain estimate is stable (needs better checks)** : could it be this one not being satisfied? I am not sure I understand the logic on how it is triggered or not.

**To Reproduce**
This behavior seem to occur sporadically.

**Expected behavior**
The EKF should not diverge when taking off and should use the Range finder data.

**Log Files and Screenshots**
https://review.px4.io/plot_app?log=18a8c251-4f93-4532-b8e8-98c5864a4d49

francelico, 2020-05-30:
Is there a way to check which source of height is used by the EKF at any point in time from the logs?

bresch, 2020-06-11:
>Is there a way to check which source of height is used by the EKF at any point in time from the logs?

Those are the bits you nee to check in `control_mode_flags`
https://github.com/PX4/Firmware/blob/a4927606ed3799361819ec7d1caaf207fe6a269f/msg/estimator_status.msg#L37-L39

priseborough, 2021-08-19:
This is an old version, the takeoff is less than 5 seconds from boot so the EKF is still settling.

closed 2021-08-19 by priseborough