PX4/PX4-ECL #835: EKF vibration induced loss of height protection when using GPS instead of Baro height

priseborough, 2020-06-02:
The detection of loss of height accuracy due to bad vibration used to more quickly reset the states to the primary height and vertical velocity sensor currently only works when Baro is used for height and GPS is used for vertical velocity. With the availability of better clipping data this should be extended to work for a bigger range of height sensors, particularly the use of GPS for height which is becoming more common.

francelico, 2020-06-04:
Hi @priseborough , do you think this could have been the cause of the behavior observed in #834 ? I was running the PX4 v1.10.1 ecl version at the time (https://github.com/PX4/ecl/tree/dec1ed0aeb395a885b60b1d61cfb57bd932c7d34). 

I was not using GPS for vertical velocity. Baro, Vision pose and Range finders were fused in the AID_MASK and EKF2_HGT_MODE was set to Baro

priseborough, 2021-08-19:
The use of clipping counter to improve robustness will have improved this behaviour, but the use of the same sensor for height and vertical velocity introduces a common mode failure that we cannot ignore.

closed 2021-08-19 by priseborough