PX4/PX4-ECL #853: Mathematical error in test_ratios computation

francelico, 2020-06-25:
Hello,

I was looking into the meaning of the EKF2_XXX_GATE parameters, in the code, the relevant line (here just for vertical velocity) is

https://github.com/PX4/ecl/blob/e4b44f704bb641fa93bedc81734c5249fcd9ed42/EKF/vel_pos_fusion.cpp#L75
 
Where if `test_ratio` is larger than 1 the measurement will be considered invalid and not fused.

Now looking at the significance of each variable in the code, I came to the conclusion that `test_ratio` is

<img src="https://render.githubusercontent.com/render/math?math=\text{test ratio} = \frac{{(\text{latest ekf state output} - \text{latest sensor measurement})}^2}{{\text{gate_size}}^2 \times \text{innov_var}}">

where

https://github.com/PX4/ecl/blob/e4b44f704bb641fa93bedc81734c5249fcd9ed42/EKF/vel_pos_fusion.cpp#L74

Where the P term is the relevant state variance entry and the R term the relevant observation variance entry (computed from the square of EKF2_XXX_NOISE or the variance message associated with the sensor measurement, whichever is higher).

Now rearranging we can see that the measurement will only be considered valid if

<img src="https://render.githubusercontent.com/render/math?math=|\text{latest ekf state output} - \text{latest sensor measurement}| \leq \text{gate size} \times \sqrt{\text{innov_var}}">

This makes sense conceptually, as I understand we are checking that the absolute difference between the current state and sensor measurement need to fit through the gate, that is dependent on X standard deviations, and that standard deviation is defined as the absolute sum (i.e. the max possible error) between the EKF state noise and the measurement noise. 

However I think this may be a mathematically incorrect way define it. We are trying to get `|noise_EKF + noise_sensor|`, but here innov_var is defined as `innov_var = noise_EKF^2 + noise_sensor^2`. For it to be correct shouldn't we define `innov_var` as 

<img src="https://render.githubusercontent.com/render/math?math=\text{innov var} = R %2B P %2B 2\sqrt{RP}">

in order to respect the `(a+b)^2 = a^2 + b^2 + 2ab` identity?

To be honest the whole EKF theory is quite complex, and I may have missed some subtlety here, so I would welcome some feedback on this.

bresch, 2020-06-26:
@francelico The variance of the sum of two independent random variables is `Var(a+b) = Var(a) + Var(b)` (see http://onlinestatbook.com/2/glossary/variance_sum_law.html). The `2ab` term is zero because they are not correlated.
For reference, the "test ratio" is often refereed as "Normalised Innovation Squared" (NIS) test and the gate is a tuning factor. See for example in: http://www.robots.ox.ac.uk/~ian/Teaching/Estimation/LectureNotes2.pdf

francelico, 2020-06-26:
Hi @bresch thanks for the explanation, it's a lot clearer now! Also appreciate the resource you linked on the NIS. Closing the issue now.

closed 2020-06-26 by francelico