PX4/PX4-ECL #867: EKF2 Unstable despite Stable Data

lelliotts20, 2020-07-14:
[image]

Hey all! 
Really struggling to understand why the EKF2 local position is wandering all over the place despite the sensor data remaining relatively stable. After about 30 seconds of sitting on the ground in simulation the pos_horiz_accuracy stats to rise and never comes back down.

Setup is as follows:
Px4 in sitl mode
Imu and mag from mavros fused using imu filter madgwick
Said data is passed through rtabmap's odometry and fed into /mavros/odometry/out
Ekf2 _aid_mask 24

lelliotts20, 2020-07-14:
Follow up with 2 logs, there seems to be two issues. One where my data causes it to go invalid in air and the other where the ekf goes invalid on the ground. (data from image got corrupted, new log isn't as crazy)
[Crashes.zip](https://github.com/PX4/ecl/files/4921460/Crashes.zip)

priseborough, 2021-08-19:
Old version of code. Significant robustness and stability improvements have been made since. Please re-open if this is still an issue with the current version.

closed 2021-08-19 by priseborough