PX4/PX4-ECL #909: EKF minimize console output

dagar, 2020-09-21:
Can we work on minimizing console output for the estimator? It gets a little absurd when running multiple (https://github.com/PX4/Firmware/pull/14650).

``` Console
INFO  [ecl/EKF] reset position to last known position
INFO  [ecl/EKF] reset velocity to zero
INFO  [ecl/EKF] reset position to last known position
INFO  [ecl/EKF] reset velocity to zero
INFO  [ecl/EKF] reset position to last known position
INFO  [ecl/EKF] reset velocity to zero
INFO  [ecl/EKF] reset position to last known position
INFO  [ecl/EKF] reset velocity to zero
INFO  [ecl/EKF] reset position to last known position
INFO  [ecl/EKF] reset velocity to zero
INFO  [ecl/EKF] reset position to last known position
INFO  [ecl/EKF] reset velocity to zero
INFO  [ecl/EKF] 11544067: EKF aligned, (baro hgt, IMU buf: 18, OBS buf: 14)
INFO  [ecl/EKF] 11644085: EKF aligned, (baro hgt, IMU buf: 18, OBS buf: 14)
INFO  [ecl/EKF] 11751436: EKF aligned, (baro hgt, IMU buf: 18, OBS buf: 14)
INFO  [ecl/EKF] 11813268: EKF aligned, (baro hgt, IMU buf: 18, OBS buf: 14)
INFO  [ecl/EKF] 11951770: EKF aligned, (baro hgt, IMU buf: 18, OBS buf: 14)
INFO  [ecl/EKF] 11973531: EKF aligned, (baro hgt, IMU buf: 18, OBS buf: 14)
```

I would propose we (incrementally) eliminate all console output by default, ensuring the equivalent information is trivial to find in the logging. Critical in flight errors and resets could still make sense if used sparingly, but even there we'll want to make sure it's again captured by logging and perhaps even handled higher level (ekf2_main) to produce a mavlink critical message rather than console noise. 

FYI @priseborough @bresch

dagar, 2020-09-28:
We also need to be careful with messages that have the potential to flood the console.

``` Console
ERROR [ecl/EKF] airspeed badly conditioned - wind covariance reset
ERROR [ecl/EKF] airspeed badly conditioned - wind covariance reset
ERROR [ecl/EKF] airspeed badly conditioned - wind covariance reset
ERROR [ecl/EKF] airspeed badly conditioned - wind covariance reset
ERROR [ecl/EKF] airspeed badly conditioned - wind covariance reset
ERROR [ecl/EKF] airspeed badly conditioned - wind covariance reset
ERROR [ecl/EKF] airspeed badly conditioned - wind covariance reset
ERROR [ecl/EKF] airspeed badly conditioned - wind covariance reset
ERROR [ecl/EKF] airspeed badly conditioned - wind covariance reset
ERROR [ecl/EKF] airspeed badly conditioned - wind covariance reset
ERROR [ecl/EKF] airspeed badly conditioned - wind covariance reset
ERROR [ecl/EKF] airspeed badly conditioned - wind covariance reset
ERROR [ecl/EKF] airspeed badly conditioned - wind covariance reset
ERROR [ecl/EKF] airspeed badly conditioned - wind covariance reset
ERROR [ecl/EKF] airspeed badly conditioned - wind covariance reset
ERROR [ecl/EKF] airspeed badly conditioned - wind covariance reset
ERROR [ecl/EKF] airspeed badly conditioned - wind covariance reset
ERROR [ecl/EKF] airspeed badly conditioned - wind covariance reset
ERROR [ecl/EKF] airspeed badly conditioned - wind covariance reset
ERROR [ecl/EKF] airspeed badly conditioned - wind covariance reset
ERROR [ecl/EKF] airspeed badly conditioned - wind covariance reset
ERROR [ecl/EKF] airspeed badly conditioned - wind covariance reset
ERROR [ecl/EKF] airspeed badly conditioned - wind covariance reset
ERROR [ecl/EKF] airspeed badly conditioned - wind covariance reset
ERROR [ecl/EKF] airspeed badly conditioned - wind covariance reset
ERROR [ecl/EKF] airspeed badly conditioned - wind covariance reset
ERROR [ecl/EKF] airspeed badly conditioned - wind covariance reset
ERROR [ecl/EKF] airspeed badly conditioned - wind covariance reset
ERROR [ecl/EKF] airspeed badly conditioned - wind covariance reset
ERROR [ecl/EKF] airspeed badly conditioned - wind covariance reset
ERROR [ecl/EKF] airspeed badly conditioned - wind covariance reset
ERROR [ecl/EKF] airspeed badly conditioned - wind covariance reset
ERROR [ecl/EKF] airspeed badly conditioned - wind covariance reset
ERROR [ecl/EKF] airspeed badly conditioned - wind covariance reset
ERROR [ecl/EKF] airspeed badly conditioned - wind covariance reset
ERROR [ecl/EKF] airspeed badly conditioned - wind covariance reset
ERROR [ecl/EKF] airspeed badly conditioned - wind covariance reset
```

priseborough, 2021-08-19:
This has been reduced, but there is scope for further improvement using data in flags. @dagar I'll keep this open for the time being.