PX4/PX4-ECL #911: EKF testing gaps

dagar, 2020-09-24:
Overall test coverage is looking good, but there are still a number of gaping holes.

[image]

 - airspeed_fusion.cpp 68.67% https://codecov.io/gh/PX4/ecl/src/master/EKF/airspeed_fusion.cpp
 - drag_fusion.cpp 0% https://codecov.io/gh/PX4/ecl/src/master/EKF/drag_fusion.cpp
 - mag_control.cpp 78.15% https://codecov.io/gh/PX4/ecl/src/master/EKF/mag_control.cpp
     - explicitly test modes other than **MAG_FUSE_TYPE_AUTO**
 - optflow_fusion.cpp 0% https://codecov.io/gh/PX4/ecl/src/master/EKF/optflow_fusion.cpp
 - sideslip_fusion.cpp 0% https://codecov.io/gh/PX4/ecl/src/master/EKF/sideslip_fusion.cpp

dagar, 2020-09-24:
Testing the different height modes is another missing piece.

bresch, 2020-09-25:
Good to know, thanks @dagar . I'll add a task for myself to remember adding basic coverage of those functions. Also, if anyone else is interested to do it, he/she's more than welcome!