PX4/PX4-ECL #916: SD card data dropout cause EKF_replay divergence

HBH-THU, 2020-10-20:
Hi everyone, I have a problem that my flight log has some data dropout, and the maximum dropout time gap is about 0.7 seconds. But when I used EKF_replay to replay the flight log, I found a serious divergence of the log. So, whether is there some solutions to compensate for this dropout, or whether is there some methods to prevent it from happening, thank you all.

CarlOlsson, 2020-10-20:
Hi,

- Get a [SanDisk Extreme U3 32GB](https://dev.px4.io/master/en/log/logging.html#sd-cards) SD card
- Increase the [log buffer](https://github.com/PX4/Firmware/blob/master/ROMFS/px4fmu_common/init.d/rcS#L37) as much as you can without running out of RAM
- Make sure CPU load is not too high (on an f4 processor < 80%)
- Only log what you actually need (First only the relevant topics, then potentially remove fields from the topics you don't need, e.g. fields from [ekf2_timestamps](https://github.com/PX4/Firmware/blob/master/msg/ekf2_timestamps.msg#L1))

Good luck!

HBH-THU, 2020-10-29:
@CarlOlsson Thanks a lot

closed 2020-10-29 by HBH-THU