PX4/PX4-ECL #934: problem of fake position fusion

garlinplus, 2020-11-11:
When there are not any horizontal aiding fusion sensor,the fake position fusion is active.
But in ecl ekf  code,last known north-east position is not update when gps or flow change state from valid to invalid.
In ecl ekf code,Only one last known position updating is found as followed:
	if (_flow_data_ready && (_imu_sample_delayed.time_us > _flow_sample_delayed.time_us - uint32_t(1e6f * _flow_sample_delayed.dt) / 2)) {
		// Fuse optical flow LOS rate observations into the main filter only if height above ground has been updated recently
		// but use a relaxed time criteria to enable it to coast through bad range finder data
		if (_control_status.flags.opt_flow && isRecent(_time_last_hagl_fuse, (uint64_t)10e6)) {
			fuseOptFlow();
			_last_known_posNE = _state.pos.xy();
		}

		_flow_data_ready = false;
	}
Actually I think _last_known_posNE should be updated in gps fusion or other horizontal sensor fusion.

garlinplus, 2020-11-11:
Actually in fake position fusion, if it's changed from other horizontal sensor fusion to none horizontal sensor fusion state, position and velocity all is reset , so the position and velocity innovation would not be so large that make ekf diverging.If it's as above thinking, there is no need to update _last_known_posNE.

bresch, 2020-11-11:
> Actually in fake position fusion, if it's changed from other horizontal sensor fusion to none horizontal sensor fusion state, position and velocity all is reset , so the position and velocity [...]

That's correct. When switching to fake fusion, the position estimate is reset to `_last_known_posNE` here:
https://github.com/PX4/PX4-ECL/blob/da9f314b69585ba169d033fec293e9ea48b9d1fd/EKF/ekf_helper.cpp#L184-L187

garlinplus, 2020-12-02:
void Ekf::controlFakePosFusion()
{
	// if we aren't doing any aiding, fake position measurements at the last known position to constrain drift
	// Coincide fake measurements with baro data for efficiency with a minimum fusion rate of 5Hz

	if (!isHorizontalAidingActive()
	    && !(_control_status.flags.fuse_aspd && _control_status.flags.fuse_beta)) {

		// We now need to use a synthetic position observation to prevent unconstrained drift of the INS states.
		_using_synthetic_position = true;

		// Fuse synthetic position observations every 200msec
		if (isTimedOut(_time_last_fake_pos, (uint64_t)2e5)) {

			// Reset position and velocity states if we re-commence this aiding method
			if (isTimedOut(_time_last_fake_pos, (uint64_t)4e5)) {
				resetHorizontalPosition();
				resetVelocity();
				_fuse_hpos_as_odom = false;

				if (_time_last_fake_pos != 0) {
					ECL_WARN_TIMESTAMPED("stopping navigation");
				}

			}
			_time_last_fake_pos = _time_last_imu;

			Vector3f fake_pos_obs_var;

			if (_control_status.flags.in_air && _control_status.flags.tilt_align) {
				fake_pos_obs_var(0) = fake_pos_obs_var(1) = sq(fmaxf(_params.pos_noaid_noise, _params.gps_pos_noise));

			} else {
				fake_pos_obs_var(0) = fake_pos_obs_var(1) = sq(0.5f);
			}

			_gps_pos_innov.xy() = Vector2f(_state.pos) - _last_known_posNE;

			// glitch protection is not required so set gate to a large value
			const Vector2f fake_pos_innov_gate(100.0f, 100.0f);

			fuseHorizontalPosition(_gps_pos_innov, fake_pos_innov_gate, fake_pos_obs_var,
						_gps_pos_innov_var, _gps_pos_test_ratio);
		}

	} else {
		_using_synthetic_position = false;
	}
}

If current horizontal sensor state change from valid to invalid,and there is no other horizontal sensor,the second if function
 isTimedOut(_time_last_fake_pos, (uint64_t)4e5) will be return true,and then the pisition will be reset to _last_known_posNE.
Actually as the name of varible _last_known_posNE,It's should be last known position.So that,should we must update _last_known_posNE when there is any horizontal sensor?

closed 2020-12-15 by bresch

bresch, 2020-12-15:
@garlinplus I fixed it in #943 , thanks!