PX4/PX4-ECL #951: EKF without baro sensor can stop fusion any height permanently

lukegluke, 2020-12-23:
I'm working with ground rover without baro, only GPS. I'm using a workaround for EKF initialization without baro (#876), but I found another issue.

If GPS data get stopped, firstly (after 5 secs) we get into [controlHeightSensorTimeouts()](https://github.com/PX4/PX4-ECL/blob/df7f261e038aa80b07a91857f2f952280938bfc6/EKF/control.cpp#L766) and trying to reset to baro, but without it we are staying on GPS height fusion (`_control_status.flags.gps_hgt` is still set).
https://github.com/PX4/PX4-ECL/blob/df7f261e038aa80b07a91857f2f952280938bfc6/EKF/control.cpp#L832-L843

Then (after 10 secs) we get into [stopGpsFusion()](https://github.com/PX4/PX4-ECL/blob/df7f261e038aa80b07a91857f2f952280938bfc6/EKF/ekf_helper.cpp#L1392) where `_control_status.flags.gps_hgt` get cleared.
That's all. Now none of the height flags is set and [controlHeightSensorTimeouts()](https://github.com/PX4/PX4-ECL/blob/df7f261e038aa80b07a91857f2f952280938bfc6/EKF/control.cpp#L766) can't do nothing and will not call [startGpsHgtFusion()](https://github.com/PX4/PX4-ECL/blob/df7f261e038aa80b07a91857f2f952280938bfc6/EKF/ekf_helper.cpp#L1223) if GPS is available again. At the same timeGpsFusion for horizontal position is returning back normally.

Additional 'else' to controlHeightSensorTimeouts() like this fix the problem for my particular case:
```
} else if (_control_status.flags.ev_hgt) {
    ...
} else {
	if (_params.vdist_sensor_type == VDIST_SENSOR_GPS) {
		// check if GPS height is available
		const gpsSample &gps_init = _gps_buffer.get_newest();
		const bool gps_hgt_accurate = (gps_init.vacc < _params.req_vacc);

		// check for inertial sensing errors in the last BADACC_PROBATION seconds
		const bool prev_bad_vert_accel = isRecent(_time_bad_vert_accel, BADACC_PROBATION);

		// reset to GPS if adequate GPS data is available and the timeout cannot be blamed on IMU data
		const bool reset_to_gps = !_gps_hgt_intermittent &&
					((gps_hgt_accurate && !prev_bad_vert_accel) || _baro_hgt_faulty);

		if (reset_to_gps) {
			// set height sensor health
			_baro_hgt_faulty = true;

			startGpsHgtFusion();

			//reset sensor offset because height can drift quite far away
			_hgt_sensor_offset = 0.0f;

			request_height_reset = true;
			failing_height_source = "";
			new_height_source = "gps";

		}
	}
}
```
I suppose that issue is related to other height sources too. Also as I see even on systems with baro there still a small probability to never return to any height fusion if baro was faulty (_baro_hgt_faulty) on time of height sensor timeout.

For your note, @dagar and for addition to working on #931. Thanks!

dagar, 2020-12-23:
FYI I kept the original baro related init changes from #931 in https://github.com/PX4/PX4-ECL/tree/pr-ekf_init_no_baro.

lukegluke, 2020-12-24:
Yes, thanks. But I added also a waiting for _gps_buffer samples same as for baro. Because of not yet cleaned checks:
https://github.com/PX4/PX4-ECL/blob/df7f261e038aa80b07a91857f2f952280938bfc6/EKF/estimator_interface.cpp#L189-L190
gps data in buffer at least 3d :)

dagar, 2020-12-24:
Yes, we should do this for all the cases as Paul described in https://github.com/PX4/PX4-ECL/pull/931#discussion_r544047491.

@lukegluke do you have a flight log testing the GPS change with no baro?

lukegluke, 2020-12-24:
> @lukegluke do you have a flight log testing the GPS change with no baro?

I don't fly just driving on ground :). If you want I can send by email links to logs I mentioned in https://github.com/PX4/PX4-ECL/issues/947#issuecomment-748083779, but I suppose it would not be very useful for you, they are recorded with older px4 and has modifications like for example lat, lon are double in deg, alt - float in m.

dagar, 2021-01-09:
> for example lat, lon are double in deg, alt - float in m.

Is this for a particular gps unit? I wouldn't mind making that change upstream.

lukegluke, 2021-01-10:
@dagar I've already suggested this here: https://github.com/PX4/PX4-Autopilot/pull/14447#issuecomment-671038104 and tried to note possible drawbacks and tasks to do. In our custom project we moved to doubles quite ago and working with general NMEA driver https://github.com/PX4/PX4-GPSDrivers/pull/58, but it is not a big deal to correct all other drivers.
But we are using just a part of px4 framework and only for rovers so can't be responsible for all boards and can't correct or make test, also as I understand QGC uses the same gps driver repository, so it also should be corrected simultaneously by someone.

I could try to make a separate patch when I have time if you need it as reference point.

priseborough, 2021-08-19:
If GPS is selected as the primary height source and it stops and there is baro data, it will continue to try and reset to GPS. Not sure what you expect EKF to do if the only source of height data stops given we do not have a no height mode available. We would need to add support for a no height mode or switch to using a 'fake' height measurement and set z and velz validity to false in the local position and global position message to stop control loops trying to use the data.

lukegluke, 2021-08-19:
> Not sure what you expect EKF to do if the only source of height data stops given we do not have a no height mode available.

I would expect that EKF without baro data try to reset to GPS if adequate GPS data is available back again.

lalawinter, 2022-07-15:
I want to use gps, imu, barometer and magnetometer to estimate the position、speed and yaw of a drone, but I don't know how to use this library