PX4/PX4-ECL #953: terrain height is not accurate

garlinplus, 2021-01-04:
When laser ranger finder is added,I find terrain height is not accurate.Even if there is not a floor,the terrain is not zero.

bresch, 2021-01-04:
@garlinplus The terrain estimator only works properly if the range finder is not used as the primary height source (`EKF2_HGT_MODE` should be baro or GPS height); otherwise, the same measurement is used to measure the altitude and the terrain elevation, which is not possible.

ashwinvk94, 2021-01-04:
The terrain estimator _terrain_vpos is used in [optical flow fusion](https://github.com/PX4/PX4-ECL/blob/master/EKF/optflow_fusion.cpp#L85). So when an optical flow sensor + rangefinder is used, the terrain estimator would be expected to work right?

garlinplus, 2021-01-05:
> e measurement is u

closed 2021-01-05 by garlinplus

garlinplus, 2021-01-05:
@bresch In the ekf update function,the terrain estimate function runTerrainEstimator is executed and  the condition whether  baro or gps is used as primary height source is neglected.So If ranger sensor is selected as the primary height source,the terrain estimator is always executed.

bool Ekf::update()
{
	bool updated = false;

	if (!_filter_initialised) {
		_filter_initialised = initialiseFilter();

		if (!_filter_initialised) {
			return false;
		}
	}

	// Only run the filter if IMU data in the buffer has been updated
	if (_imu_updated) {
		// perform state and covariance prediction for the main filter
		predictState();
		predictCovariance();

		// control fusion of observation data
		controlFusionModes();

		// run a separate filter for terrain estimation
		runTerrainEstimator();

		updated = true;

		// run EKF-GSF yaw estimator
		runYawEKFGSF();
	}

	// the output observer always runs
	// Use full rate IMU data at the current time horizon
	calculateOutputStates();

	return updated;
}

void Ekf::runTerrainEstimator()
{
	// If we are on ground, store the local position and time to use as a reference
	if (!_control_status.flags.in_air) {
		_last_on_ground_posD = _state.pos(2);
	}

	// Perform initialisation check and
	// on ground, continuously reset the terrain estimator
	if (!_terrain_initialised || !_control_status.flags.in_air) {
		_terrain_initialised = initHagl();

	} else {

		// predict the state variance growth where the state is the vertical position of the terrain underneath the vehicle

		// process noise due to errors in vehicle height estimate
		_terrain_var += sq(_imu_sample_delayed.delta_vel_dt * _params.terrain_p_noise);

		// process noise due to terrain gradient
		_terrain_var += sq(_imu_sample_delayed.delta_vel_dt * _params.terrain_gradient)
				* (sq(_state.vel(0)) + sq(_state.vel(1)));

		// limit the variance to prevent it becoming badly conditioned
		_terrain_var = math::constrain(_terrain_var, 0.0f, 1e4f);

		// Fuse range finder data if available
		if (_range_sensor.isDataHealthy()) {
			fuseHagl();
		}

		if (_flow_for_terrain_data_ready) {
			fuseFlowForTerrain();
			_flow_for_terrain_data_ready = false;
		}

		// constrain _terrain_vpos to be a minimum of _params.rng_gnd_clearance larger than _state.pos(2)
		if (_terrain_vpos - _state.pos(2) < _params.rng_gnd_clearance) {
			_terrain_vpos = _params.rng_gnd_clearance + _state.pos(2);
		}
	}

	updateTerrainValidity();
}

garlinplus, 2021-01-05:
> The terrain estimator _terrain_vpos is used in [optical flow fusion](https://github.com/PX4/PX4-ECL/blob/master/EKF/optflow_fusion.cpp#L85). So when an optical flow sensor + rangefinder is used, the terrain estimator would be expected to work right?
Actually I think if you want to use optical flow sensor and ranger finder,you can get los rate wx and wy from flow,and get distance r from ranger.Then we can get body velocity vbx and vby as followed:
vbx = sign * wx * r
vby = sign * wy * r
The sign varible may be -1 or +1,it's dependent on flow orientation.Then we can rotate the body velocity to ned local frame,And fuse it with the same function that is used to fuse gps velocity.As this way,the flow fusion is simpler  and simple.

ashwinvk94, 2021-01-05:
Thanks, I will look into using the gps velocity plugin. I understood the body velocity estimation using optical flow. But when using the gps velocity plugin can I send velocities in the body frame? Or does it have to be in the NED frame?

garlinplus, 2021-01-14:
> e same measurement is used to mea

hi,@bresch.The terrain estimator of PX4 EKF2 doesn't consider wheather the primary height source is ranger.
If there is a ranger,the terrain estimator is running.

ashwinvk94, 2021-01-14:
Yes but both the height and terrain estimator will use the ranger finder, which causes problems. One that I know of is that the optical flow fusion uses the [difference of the two](https://github.com/PX4/PX4-ECL/blob/master/EKF/optflow_fusion.cpp#L85). Which I still don't understand.

bresch, 2021-01-15:
> Yes but both the height and terrain estimator will use the ranger finder, which causes problems. One that I know of is that the optical flow fusion uses the difference of the two. Which I still don't understand.

Yes, using range height as primary height source and flow does not seem to work properly at the moment. As a quick workaround, you could disable terrain height estimation; or if you have a good idea how to solve it nicely, please submit your solution in a PR and we can discuss it.