PX4/PX4-ECL #955: Terrain height + z-position for optical flow fusion

ashwinvk94, 2021-01-04:
The _**height above ground range**_ used for optical flow fusion is set as[ _terrain_vpos - z_position.](https://github.com/PX4/PX4-ECL/blob/master/EKF/optflow_fusion.cpp#L85) But when I am using a rangefinder(`EKF2_HGT_MODE` is set rangefinder), the z position is directly measuring the height above the ground. In this situation, shouldn't the heightAboveGndEst be equal to _state.pos(2) or z_position?

bresch, 2021-01-05:
Even in range finder primary height mode, the inertial data is used and the vertical GPS velocity is fused (if available). This means that even if the terrain height changes, the altitude of the EKF (`state.pos(2)`) will be stabilized by the other sensors and a relative terrain estimate can be obtained.

ashwinvk94, 2021-01-07:
This issue was solved by the following steps:

1. `EKF2_HGT_MODE` was set to rangefinder
2. barometer aiding was removed by commenting out the respective [lines](https://github.com/PX4/PX4-ECL/blob/03fed323ab950125114eb34309d3a79119e572ee/EKF/control.cpp#L992-L995) from the controlHeightFusion() in control.cpp. This ensured that the rangefinder would solely be used to estimate the the altitude of the EKF (`state.pos(2)`)

closed 2021-01-07 by ashwinvk94