PX4/PX4-ECL #980: Rotate yaw in position control mode,the horizontal velocity is wrong with gps measured velocity.

garlinplus, 2021-02-18:
When I rotate yaw and push pitch RC channel make copter do circular motion.In circular motion process,the copter action is correct.When I realse the RC scroller.the copter immediately return to stablized attitude,and after 2s,the copter always fly away with a permant yaw angle.
[image]

garlinplus, 2021-02-18:
the log address is https://logs.px4.io/plot_app?log=912ad7f9-74e4-45c1-9a74-5a6072930797

CarlOlsson, 2021-02-18:
That's a very interesting logfile, the IMU biases diverge during the rotation. Do you think you could collect a replay logfile during the same maneuver?

garlinplus, 2021-02-18:
> he IMU biases diverge during the rotation

[image]
[image]
The gyro and acc bias seems to be large than formal flying log as followed:
[image]
[image]

garlinplus, 2021-02-18:
> That's a very interesting logfile, the IMU biases diverge during the rotation. Do you think you could collect a replay logfile during the same maneuver?

The imu bias is larger than formal flying,but Even if as that,the gps and mag measure can not correct those with ekf measure update process?

abishekvupputur, 2021-04-01:
I notice that your yaw angular rate is about 100 deg/s. I think it is quite possible that you are measuring centrifugal acceleration in the body reference frame. Due to lack of observability of centrifugal force, in body frame it is interpreted like a "added bias (function of spin) during large yaw rates". So it makes sense that the bias estimates is diverging to accommodate for this error caused as a consequence of spin. I think the EKF is working perfectly to accommodate this. There 2 possible ways to get rid of this problem.
1. Centre your IMU closer to the center of gravity of the drone. The closer you are, lower is your "acceleration bias divergence".
2. Write a small calibration routine, manually calculate the offset of imu from the centre of gravity and calculate your acceleration caused as a result of spin before passing the values through your EKF. This compensation for the added acceleration can be done by subtracting a_c from the measured acceleration from IMU:
>> a_c = ω x ω x r.
where,
a_c - calculated acceleration in [x,y,z]
ω - body angular velocity in [x,y,z] (gyro bias corrected values)
r - imu offset from cog in [x,y,z]

May be in future, we can also have a IMU offset parameter to account for this error. Or extend the EKF to include IMU offset prediction based on body angular rate to measured acceleration.

abishekvupputur, 2021-04-01:
[](url)https://github.com/PX4/PX4-ECL/issues/621

priseborough, 2021-08-24:
This looks like the issue that was fixed by https://github.com/PX4/PX4-ECL/pull/1026

Please re-open if not fixed for you.

closed 2021-08-24 by priseborough