PX4/PX4-ECL #982: EKF2 update mag bias function problem

garlinplus, 2021-02-24:
The function as followed:
void Ekf2::update_mag_bias(Param &mag_bias_param, int axis_index)
{
	if (_valid_cal_available[axis_index]) {

		// calculate weighting using ratio of variances and update stored bias values
		const float weighting = constrain(_param_ekf2_magb_vref.get() / (_param_ekf2_magb_vref.get() +
						  _last_valid_variance[axis_index]), 0.0f, _param_ekf2_magb_k.get());
		const float mag_bias_saved = mag_bias_param.get();

		_last_valid_mag_cal[axis_index] = weighting * _last_valid_mag_cal[axis_index] + mag_bias_saved;

		mag_bias_param.set(_last_valid_mag_cal[axis_index]);
		mag_bias_param.commit_no_notification();

		_valid_cal_available[axis_index] = false;
	}
}

Should it be as followed:
_last_valid_mag_cal[axis_index] = weighting * _last_valid_mag_cal[axis_index] + (1 - weighting ) *mag_bias_saved;

bresch, 2021-02-24:
No, because `_last_valid_mag_cal` is the "bias increment" (i.e.: `new_bias - old_bias`). 
So 
```
w * (new_bias - old_bias) + old_bias 
= w * new_bias - w*old_bias + old_bias 
= w * new_bias + (1-w) * old_bias
```

closed 2021-02-24 by bresch

garlinplus, 2021-02-25:
> No, because `_last_valid_mag_cal` is the "bias increment" (i.e.: `new_bias - old_bias`).
> So
> 
> ```
> w * (new_bias - old_bias) + old_bias 
> = w * new_bias - w*old_bias + old_bias 
> = w * new_bias + (1-w) * old_bias
> ```

But In desciption of Ekf2 state,the mag bias is only a bias,and the _last_valid_mag_cal is updated as followed:
				// Start checking mag bias estimates when we have accumulated sufficient calibration time
				if (_total_cal_time_us > 120_s) {
					// we have sufficient accumulated valid flight time to form a reliable bias estimate
					// check that the state variance for each axis is within a range indicating filter convergence
					const float max_var_allowed = 100.0f * _param_ekf2_magb_vref.get();
					const float min_var_allowed = 0.01f * _param_ekf2_magb_vref.get();

					// Declare all bias estimates invalid if any variances are out of range
					bool all_estimates_invalid = false;

					for (uint8_t axis_index = 0; axis_index <= 2; axis_index++) {
						if (status.covariances[axis_index + 19] < min_var_allowed
						    || status.covariances[axis_index + 19] > max_var_allowed) {
							all_estimates_invalid = true;
						}
					}

					// Store valid estimates and their associated variances
					if (!all_estimates_invalid) {
						for (uint8_t axis_index = 0; axis_index <= 2; axis_index++) {
							_last_valid_mag_cal[axis_index] = status.states[axis_index + 19];
							_valid_cal_available[axis_index] = true;
							_last_valid_variance[axis_index] = status.covariances[axis_index + 19];
						}
					}
				} 

As above _last_valid_mag_cal is equal with states[19],states[20] and states[21].These states is mag bias in body frame,and it's not increment.

bresch, 2021-02-25:
> As above _last_valid_mag_cal is equal with states[19],states[20] and states[21].These states is mag bias in body frame,and it's not increment.

That's right, but the calibration is already applied on the sensor data, before being sent in the EKF. It means that the bias estimate is on top of the existing calibration.