PX4/PX4-ECL #984: ekf2 tilt align cost time too long about 7s

garlinplus, 2021-02-27:
ekf2 tilt align cost time too long about 7s.
Is the init covariance too big wo make it's angle error covergente slowly? 	
// monitor the tilt alignment
	if (!_control_status.flags.tilt_align) {
		// whilst we are aligning the tilt, monitor the variances
		const Vector3f angle_err_var_vec = calcRotVecVariances();

//		_angle_err_var_counter++;
//
//		if(_angle_err_var_time == 0){
//			_angle_err_var_time = hrt_absolute_time();
//		}
		// Once the tilt variances have reduced to equivalent of 3deg uncertainty, re-set the yaw and magnetic field states
		// and declare the tilt alignment complete
		if ((angle_err_var_vec(0) + angle_err_var_vec(1)) < sq(math::radians(3.0f))) {

//			printf("var counter:%d,%lld.\n",_angle_err_var_counter,hrt_elapsed_time(&_angle_err_var_time));

			_control_status.flags.tilt_align = true;
			_control_status.flags.yaw_align = resetMagHeading(_mag_lpf.getState()); // TODO: is this needed?

			// send alignment status message to the console
			const char* height_source = nullptr;
			if (_control_status.flags.baro_hgt) {
				height_source = "baro";

			} else if (_control_status.flags.ev_hgt) {
				height_source = "ev";

			} else if (_control_status.flags.gps_hgt) {
				height_source = "gps";

			} else if (_control_status.flags.rng_hgt) {
				height_source = "rng";

			} else {
				height_source = "unknown";

			}
			if(height_source){
				ECL_INFO("%llu: EKF aligned, (%s hgt, IMU buf: %i, OBS buf: %i)",
					(unsigned long long)_imu_sample_delayed.time_us, height_source, (int)_imu_buffer_length, (int)_obs_buffer_length);
			}
		}
	}

bresch, 2021-03-01:
I don't think the initial quaternion covariance is too large, but it is large enough to represent the tilt accuracy after the coarse alignment made using the accelerometer (including some potential sensor bias). The fine alignment is performed using a "quasi-stationary alignment" technique by fusing a synthetic (i.e.:"fake") position update with a small error (std dev of 0.5m). It takes some time (almost 7 seconds as you said) because everything is tuned to be robust to work on all platform and all hardware without issues.
The fine alignment time can be reduced by tailoring the tuning to your own setup and use case. We're also planning to add a "Zero Velocity Update (aka: ZVU)" logic to speed up the initial convergence of the estimator in stationary conditions.

garlinplus, 2021-03-02:
> aternion covariance is large enough to repr
In reality, ekf2 has a fake position fusion.It assume robot is not moving to estimate attitude.
can it be used to replace the "zero velocity update" to accelerate the tilt time?

garlinplus, 2021-03-02:
> The initial quaternion covariance is large enough to represent the tilt accuracy after the coarse alignment made using the accelerometer (including some potential sensor bias). The fine alignment is performed using a "quasi-stationary alignment" technique by fusing a synthetic (i.e.:"fake") position update with a small error (std dev of 0.5m). It takes some time (almost 7 seconds as you said) because everything is tuned to be robust to work on all platform and all hardware without issues.
> The fine alignment time can be reduced by tailoring the tuning to your own setup and use case. We're also planning to add a "Zero Velocity Update (aka: ZVU)" logic to speed up the initial convergence of the estimator in stationary conditions.

I tuned initialized parameters and it tilt error only need 4s to convergente.

bresch, 2021-03-02:
If you know that the drone won't move during boot, you can reduce the measurement noise of the fake position fusion, this would also reduce the tilt alignment convergence time: https://github.com/PX4/PX4-ECL/blob/098d5ce5c034f630b1b440a02ba78a13d092b8c7/EKF/control.cpp#L1308

garlinplus, 2021-03-03:
Yes,you are right!

closed 2021-03-04 by priseborough