PX4/PX4-ECL #985: some think about ekf2 framework

garlinplus, 2021-03-02:
In current ekf2 framework, ekf estimator run in internal layer,and complementary filter run in extern layer.
Can we use two thread to run these,one thread for complmentary filter and one for ekf estimator.
The complementary thread is used to add sensor data to ringbuffer,and get data from every buffer for sensors to ekf thread.
In this way,the sensor data can be immediately processed and not omited,and ekf estiamtor maybe can run in a high frequncy.

dagar, 2021-03-05:
I've actually already prototyped this and I think it makes sense for PX4 architecturally. I'd like to discuss the 

 - we move the output predictor to a new module that runs directly off of raw sensor data
 - the new module handles running the output predictor, filtering control data, and publishing one kinematically consistent message with all data the controllers need (angular velocity, acceleration, velocity, etc)
 - ecl/EKF can now run synchronized with integrated IMU updates rather than have its own hardcoded update period and have to downsample
 - the logging requirements for ekf2 replay are now drastically reduced (200 Hz -> 100 Hz)

Thoughts?

garlinplus, 2021-03-08:
It sounds good!