PX4/PX4-ECL #997: False negative unit test result for visionHeightFusion

ecmnet, 2021-03-31:
I noticed that the unit test regarding VisionHeightFusion fails. After vision stop is triggered, the test procedure checks with
`EXPECT_TRUE(_ekf_wrapper.isIntendingVisionHeightFusion())` whether `isIntendingVisionHeightFusion()` returns true. As vision has stopped height mode can never be in ev_hgt mode (fallback to range or baro). IMO the check needs to be `EXPECT_FALSE(_ekf_wrapper.isIntendingVisionHeightFusion())`.

bresch, 2021-03-31:
Are you talking about this test?
https://github.com/PX4/PX4-ECL/blob/da06f252166d0528624c3954b97e9e658ddf91e3/test/test_EKF_fusionLogic.cpp#L381-L383
Have you seen the two "TODO"? :D

ecmnet, 2021-03-31:
Yes, I've seen those :D - my contribution was just to clarify shortly what has to be done from my point of view.

closed 2021-03-31 by ecmnet