PX4/PX4-ECL #1002: flow fusion of ekf2

garlinplus, 2021-04-19:
EKF2 position is drift when  Flow only is used  for velocity fusion.How to settle this?

bresch, 2021-04-19:
Is the velocity drifting or just the position? It is normal that the position is slowly drifting because optical flow only provides velocity aiding. Is it drifting a lot? By how much? Do you have a log?

garlinplus, 2021-06-03:
> city drifting or just the position? It is normal that the position is slowly drifting because optical flow only provides velocity aiding. Is it drifting a lot? By how much? Do you have a log?
Flow velocity can be used to caculate a position measure to do position fusion.
This method drift slower than normal method.

bresch, 2021-06-07:
I don't see why integrating the velocity (from flow) to position and then fuse it is better than directly fusing the flow data. If it gives better results for you, that's probably because the tuning is different or that something is wrong...

priseborough, 2021-08-24:
Some users have achieved low drift rates with the exisiting code with re-tuning with smaller observation noise parameters.

closed 2021-08-24 by priseborough