PX4/PX4-ECL #1008: Constant drift when EV frequency drops below a certain threshold

canberkgurel, 2021-05-07:
Here is my data flow:

```
Gazebo -- ground_truth/odometry --> gt2vision -- camera/odom/sample --> odom_throttler -- camera/odom/sample_throttled --> px4_realsense_bridge -- mavros/odometry/out --> EKF
```

Gazebo provides the ground truth in /map frame, at ~90 Hz. There are two intermediate nodes that operate on the ground truth odometry. Due to a reason that I don't know yet, while the drone is flying the frequency of the mavros/odometry/out topic drops to ~10 Hz. (Why this happens is not the reason I'm creating this issue, it's something I'll address separately.) 

The problem is that when the odometry frequency (that the FCU is receiving) is low (e.g., 10 Hz) the EKF doesn't throw any errors nor PX4 tries to land the drone but rather the EKF outputs bogus estimates which causes the drone to drift away very slowly. So, the drone is no longer able to hold its position but it's drifting away. I've repeated this experiment many times, the drone always drifts in the same direction (front-left). At first, I suspected that something was wrong with my TF-tree but I don't think that's what the problem is, the drone is able to take off and fly for some time (20-30 seconds). The drift starts happening only after the frequency drops to ~10 Hz.

I think this is a bug that needs to be handled.

dagar, 2021-05-07:
Is the configured delay (EKF2_EV_DELAY) still correct when your odometry drops to 10 Hz? How do you know whatever's causing the rate change isn't also introducing more latency in the transmission?

canberkgurel, 2021-05-07:
@dagar I fixed the problem with the frequency drop - the problem was related to the misconfigured [topic_tools/drop](http://wiki.ros.org/topic_tools/drop). Now `mavros/odometry/out` is constant at 30 Hz. However, I'm still experiencing the drift in the same direction. My `EKF2_EV_DELAY` was set to the default value (175 ms). As I'm obtaining the odometry from the simulation environment, I expect the delay to be minimal. I set the `EKF2_EV_DELAY = 35 ms`. Still the same problem. I'll continue troubleshooting on Monday; I'll get to the bottom of this.

canberkgurel, 2021-05-10:
@dagar I found the problem. I did a `rostopic echo /mavros/setpoint_raw/local` and found out that the drift starts when transition from `coordinate_frame: 1, type_mask: 4088` to `coordinate_frame: 8, type_mask: 4039` happens (copied full message below). I checked the `setpoint_raw.cpp` and it looks like both coordinate frames 1 (local NED) and 8 (body NED) are supported. So something in this callback function is causing the behavior I'm seeing, let me know what you think. https://github.com/mavlink/mavros/blob/c8e166e0654242f5a0bb901aee2195ae0f21077e/mavros/src/plugins/setpoint_raw.cpp#L155

```
header: 
  seq: 2316
  stamp: 
    secs: 77
    nsecs:  32000000
  frame_id: ''
coordinate_frame: 1
type_mask: 4088
position: 
  x: -0.00619426509365
  y: -0.00134289637208
  z: 0.998100304976
velocity: 
  x: 0.0
  y: 0.0
  z: 0.0
acceleration_or_force: 
  x: 0.0
  y: 0.0
  z: 0.0
yaw: 0.0
yaw_rate: 0.0
---
header: 
  seq: 2317
  stamp: 
    secs: 77
    nsecs:  56000000
  frame_id: ''
coordinate_frame: 8
type_mask: 4039
position: 
  x: 0.0
  y: 0.0
  z: 0.0
velocity: 
  x: 0.0
  y: 0.0
  z: 0.0
acceleration_or_force: 
  x: 0.0
  y: 0.0
  z: 0.0
yaw: 0.0
yaw_rate: 0.0
```

closed 2021-08-24 by priseborough