PX4/PX4-ECL #1021: EKF mathematical equations

NishanthARao, 2021-07-01:
First of all, thanks for providing an efficient, light-weight implementation for the EKF. They have been really useful.

However, I would like to know if you have any references like journals / conference papers etc., where we users in the research community can cite them / include the mathematical equations of the EKF implementation in your library. This way, it will help us accurately quantify the sensor fusion technique, rather than using the standard kalman filter equations, and then the implementation that we end up using is totally different.

Thanks in advance!

bresch, 2021-07-13:
The derivation for the EKF is done using the python symbolic toolbox "sympy" here: https://github.com/PX4/PX4-ECL/blob/8a0510d52e1df9589cb37b379a98ca4931eccb20/EKF/python/ekf_derivation/main.py#L575-L596 
As you can see, the model is quite straightforward and the EKF equations (covariance prediction, etc.) are standard. 
The derivation is then optimized using the `cse` (_Common Subexpression Elimination_) function to finally produce some efficient C code that can be used in the implementation.

NishanthARao, 2021-07-14:
I see. So for the sensor fusion from multiple sensors, you just have multiple state update equations right?

Anyways, thanks a lot for your response!

closed 2021-07-14 by NishanthARao

bresch, 2021-07-14:
> So for the sensor fusion from multiple sensors, you just have multiple state update equations right?

Exactly, the measurements are fused asynchronously and separately ("sequential update").