PX4/PX4-ECL #1025: EKF estimates diverge from the Rangefinder and EV inputs in Z-axis

canberkgurel, 2021-07-08:
The `EKF2_AID_MASK` is set to 24, `EKF2_HGT_MODE` is set to Range sensor.

The plot below shows what the problem is. Given what the `EKF2_AID_MASK` & `EKF2_HGT_MODE` are set to, the EKF must be using the rangefinder and the EV estimates (along with the FCU IMU) to generate the local_position/pose, which is diverging from the two inputs in Z-axis as the drone flies. The controller uses the local_position/pose to reach the given setpoints, as you'd expect, I'm observing a severe amount of drift in the altitude axis. In particular, the drone is too high between 15-50 and too low between 80-90 on the time axis. I ran this experiment many times (>35) and I'm seeing a similar pattern in every flight. Please find the attached ulg file.

@dagar @julianoes Any help is appreciated.

[image]

[05cf0242-6e90-446d-8099-83d617e61a3e.zip](https://github.com/PX4/PX4-ECL/files/6785532/05cf0242-6e90-446d-8099-83d617e61a3e.zip)

dagar, 2021-07-08:
Uploaded to flight review for convenience. https://logs.px4.io/plot_app?log=c8fb4085-d231-41b8-9a2d-5d09b2f99881

Would you be able to try reproducing on v1.12? Quite a lot of work (and testing) has gone into ekf2 (ecl/EKF) since v1.11.

canberkgurel, 2021-07-08:
@dagar Thanks! That's good to know that there have been some improvements in the ecl/EKF. I know you've been working on creating a stable v1.12 release, last time I checked it was at the release candidate stage, I was planning to wait until the stable v1.12 release. Would you recommend the v1.12.0-rc1 for this experiment?

garlinplus, 2021-07-26:
[image]
From this log,the terrain estimation may be not accurate.

garlinplus, 2021-07-26:
And the height sensor is baro,you can select ranger or ev as height estimation sensor.Terrain estimation is depending on baro and ranger.If baro has a drift,the terrain will be not occurate.

canberkgurel, 2021-07-28:
@garlinplus Hi! The _EKF2_HGT_MODE_ is set to Rangefinder, and I have a Rangefinder that's plugged into the FCU. I confirmed that the data is coming thru by monitoring the distance measurements using the Mavlink Inspector. Are you saying that the drone wasn't using the Rangefinder, but rather it was switching to the Barometer?

priseborough, 2021-08-24:
This log indicates an old version of the code - what version of the ecl EKF are you using?

canberkgurel, 2021-08-24:
@priseborough I believe the version of ecl EKF is ecl @ ec93490. I found the last PX4-Autopilot commit that the drone had and then looked up the corresponding ecl EKF commit.

Here is the commit link: https://github.com/PX4/ecl/tree/ec934908900b23ee273d1a9f82364b7b38423200

m1baldwin, 2021-10-07:
Rangefinder is also showing initial condition of about 10-13cm estimated as the height of the sensor off the ground. This platform is the px4vision dev kit with a leddar One module used instead of the default altitude sensor that comes with the platform. Just wondering if this can be causing a problem? 

Also, plots of the drone's estimated z-position compared to the z-setpoint shows the internal estimator and controls think the drone is achieving the correct behavior, but the rangefinder is showing that there is actually a "slope" to the height the drone has throughout the mission